gnea/grbl #97: One step segment is lost when ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING is not enabled.

usbcnc, 2017-01-19:
I have tried the following g-code(8.nc), this Y axis will not go back to 0(400steps/mm). I then simplify the code to test.nc. Finally I found if the segment has only one step, then the step is not executed and this lead the stepper and planner out of sync.
Here is why
          st_prep_block->step_event_count = pl_block->step_event_count;
here st_prep_block->step_event_count is equal to 1,
        st.counter_x = st.counter_y = st.counter_z = (st.exec_block->step_event_count >> 1);
will give st.counter_x ,st.counter_y,st.counter_z all to 0.

Since there is only one step 
  if (st.counter_y > st.exec_block->step_event_count) {
    st.step_outbits |= (1<<Y_STEP_BIT);
    st.counter_y -= st.exec_block->step_event_count;
    if (st.exec_block->direction_bits & (1<<Y_DIRECTION_BIT)) { sys_position[Y_AXIS]--; }
    else { sys_position[Y_AXIS]++; }
  }
will not go true and the step is skipped.

I change from 
  if (st.counter_y > st.exec_block->step_event_count) {
to
  if (st.counter_y >= st.exec_block->step_event_count) {
The problem is solved, however I am not sure if this is the correct way.

I tried to double the st.exec_block->step_event_count so each one will be even number, but that does not work. 
Sonny may have a better solution to this. This issue also exist in 0.9 code.

[test.zip](https://github.com/gnea/grbl/files/715705/test.zip)
[8.zip](https://github.com/gnea/grbl/files/715706/8.zip)

usbcnc, 2017-01-19:
Make the following change also fix the issue.
        #ifndef ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING
          for (idx=0; idx<N_AXIS; idx++) 
	{ 
		st_prep_block->steps[idx] = pl_block->steps[idx] << 1;
       }
          st_prep_block->step_event_count = pl_block->step_event_count << 1;

chamnit, 2017-01-19:
@usbcnc : I would just do a greater than one check to compute `st.counter_x = st.counter_y = st.counter_z = (st.exec_block->step_event_count >> 1);`. If it's 1, just set to `st.counter_x = st.counter_y = st.counter_z = 1;`.

chamnit, 2017-01-19:
@usbcnc : Nevermind. That breaks the Bresenham algorithm. Your method is fine.

quangvi, 2017-01-19:
thank all people, i had change my code -> lost step had solve, if ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING is not enabled , and GRBL post to ARM , (timer prescaler is 16bit) . Now, AMASS Vs Not AMASS . what is better ?

usbcnc, 2017-01-21:
@chamnit , In my case, I am not doing extremely slow. It only if the segment of one step will show the issue.
The reason I am disabling AMASS is I am experiment constant acceleration as well as S-curve, For the time being I need to disable AMASS, I think constant acceleration can co-exist with AMASS,  Also I am experiment the "perfect" arc on the GRBL, those are the features I like like to try. 
 found sometime the program never go back to 0 leave a few steps off . By carefully analyze the g-code I am sending to and the code path eventually lead to find this issue. 
BTW, I used the Windows simulation to catch this issue. 
I am not arduino guy, I am always running GRBL on STM32F103 board. Maybe I will think of STM32F4 sometime since I find some good feature of STM32F4.

chamnit, 2017-01-21:
@usbcnc : I did not say there isn't a problem isn't there or that I won't be fixing it. I said that this problem has never been brought up because it should be exceedingly rare. Mainly because AMASS is enabled by default, and this bug has not surfaced in large part due to how AMASS works. 

I checked over the stepper code and there shouldn't be a problem altering the steps and step_event_count. Below are the changes I would make. Could you apply these changes and test it? For both, AMASS enabled and disabled, because it ensures both are fixed. 

Line 347:
```
        st.counter_x = st.counter_y = st.counter_z = st.exec_block->step_event_count;
```

Line 635:
```
          for (idx=0; idx<N_AXIS; idx++) { st_prep_block->steps[idx] = (pl_block->steps[idx] << 1); }
```

Line 641:
```
          for (idx=0; idx<N_AXIS; idx++) { st_prep_block->steps[idx] = (pl_block->steps[idx] << (MAX_AMASS_LEVEL+1)); }
```

chamnit, 2017-01-21:
@usbcnc : Whoops. Made an error. Your original proposed change should be right. Just make this change to make sure AMASS is applied this fix as well. 

Lines 641-642:
```
          for (idx=0; idx<N_AXIS; idx++) { st_prep_block->steps[idx] = (pl_block->steps[idx] << (MAX_AMASS_LEVEL+1)); }
          st_prep_block->step_event_count = pl_block->step_event_count << (MAX_AMASS_LEVEL+1);
```

usbcnc, 2017-01-22:
@chamnit. The initial bug exist when AMASS is disabled. Nothing to do with feed rate or acceleration.

To change Line 641-642 it works for AMASS enabled.(Do we need to do it? It is already scaled up).

My change of 
          for (idx=0; idx<N_AXIS; idx++) { st_prep_block->steps[idx] = pl_block->steps[idx] << 1; }
          st_prep_block->step_event_count = pl_block->step_event_count << 1;
works well.

However I modified these lines and still work, I prefer these lines to be in. What do you think? (they work for (AMASS enabled or not).
  if (st.counter_x >= st.exec_block->step_event_count) {
  if (st.counter_y >= st.exec_block->step_event_count) {
  if (st.counter_z >= st.exec_block->step_event_count) {

Original code was 
  if (st.counter_x > st.exec_block->step_event_count) 
  if (st.counter_y > st.exec_block->step_event_count) 
  if (st.counter_z > st.exec_block->step_event_count) 

counter_ value was from 1 to st.exec_block->step_event_count.
Now its value becomes from 0 to st.exec_block->step_event_count -1.

It surely changes the steps in certain case, but I tend to believe the one with = is another option. What do you think?

chamnit, 2017-01-22:
@usbcnc : Now that I think about it. You're right. It doesn't have anything to do with feed or acceleration. This would only occur if a g-code block commands a motion that moves only one step and AMASS is disabled, which is only there for machines that need to move slower than 1mm/min. So, this is even rarer than I originally thought. I'm categorizing this a non-issue and closing this.

I'll still apply your fix with AMASS disabled. When AMASS enabled, I'll do a compile-time check instead of the fix to prevent the AMASS level from being zero. When it's zero, it'll exhibit this same problem.

closed 2017-01-22 by chamnit