gnea/grbl #100: Getting error:2 with LaserWeb

Protoneer, 2017-01-21:
I am trying to setup LaserWeb3 but I get a error:2 after statup.

from https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages

I can see it indicates a "Numeric value format is not valid or missing an expected value." issue
but I am not sure how to resolve it?

I have already tried to reset to defaults with no luck.

tbfleming, 2017-01-21:
This sounds like data loss. What usb-serial chip does your board have?

Protoneer, 2017-01-21:
No usb-serial chip. ATmega to Raspberry Pi serial pins. I wonder if there
is a way for LaserWeb3 to show the raw data...

tbfleming, 2017-01-21:
server.js shows its side on the console. The two chips have incompatible IO voltages. Are you using a 3.3v - 5v level shifter?

Protoneer, 2017-01-21:
I am running this on my Raspberry CNC Hat that has been working fine with
V0.9.

My LaserWeb3 Console reports:

***************************************************************
                 ---- LaserWeb Started ----
***************************************************************
  Access the LaserWeb User Interface:
  1. Open Chrome
  2. Go to :  http://127.0.1.1:8000/
***************************************************************

* Updates:
  Remember to check the commit log on
  https://github.com/LaserWeb/LaserWeb3/commits/master
  regularly, to know about updates and fixes, and then when ready
  update LaserWeb3 accordingly by runninggit pull

* Support:
  If you need help / support, come over to
  https://plus.google.com/communities/115879488566665599508
***************************************************************

WARN:Connecting to Port /dev/ttyAMA0,115200
WARN:Connected to /dev/ttyAMA0 at 115200
Recv: error:2
Recv: error:2

Protoneer, 2017-01-21:
bCNC works just fine...

On Sun, Jan 22, 2017 at 8:56 AM, Kruger Creations <krugercreations@gmail.com

> I am running this on my Raspberry CNC Hat that has been working fine with
> V0.9.
>
> My LaserWeb3 Console reports:
>
>
> ***************************************************************
>                  ---- LaserWeb Started ----
> ***************************************************************
>   Access the LaserWeb User Interface:
>   1. Open Chrome
>   2. Go to :  http://127.0.1.1:8000/
> ***************************************************************
>
>
> * Updates:
>   Remember to check the commit log on
>   https://github.com/LaserWeb/LaserWeb3/commits/master
>   regularly, to know about updates and fixes, and then when ready
>   update LaserWeb3 accordingly by runninggit pull
>
> * Support:
>   If you need help / support, come over to
>   https://plus.google.com/communities/115879488566665599508
> ***************************************************************
>
> WARN:Connecting to Port /dev/ttyAMA0,115200
> WARN:Connected to /dev/ttyAMA0 at 115200
> Recv: error:2
> Recv: error:2
>
>
>
>
>
>
>
>> server.js shows its side on the console. The two chips have incompatible
>> IO voltages. Are you using a 3.3v - 5v level shifter?
>>
>> <https://github.com/gnea/grbl/issues/100#issuecomment-274283939>, or mute
>> the thread
>>
>
>
>
> Kind regards,
>
> *Bertus Kruger*
> *Mobile - 021 078 5392*

Protoneer, 2017-01-21:
Right....

I found the issue. LaserWeb assumes its a USB connection that always resets
the Arduino on connect. Once I start LaserWeb3 and press the reset button
on the board it picks up GRBL.  I will report it as a LaserWeb3 issue.

On Sun, Jan 22, 2017 at 8:57 AM, Kruger Creations <krugercreations@gmail.com

> bCNC works just fine...
>
> On Sun, Jan 22, 2017 at 8:56 AM, Kruger Creations <
>
>> I am running this on my Raspberry CNC Hat that has been working fine with
>> V0.9.
>>
>> My LaserWeb3 Console reports:
>>
>>
>> ***************************************************************
>>                  ---- LaserWeb Started ----
>> ***************************************************************
>>   Access the LaserWeb User Interface:
>>   1. Open Chrome
>>   2. Go to :  http://127.0.1.1:8000/
>> ***************************************************************
>>
>>
>> * Updates:
>>   Remember to check the commit log on
>>   https://github.com/LaserWeb/LaserWeb3/commits/master
>>   regularly, to know about updates and fixes, and then when ready
>>   update LaserWeb3 accordingly by runninggit pull
>>
>> * Support:
>>   If you need help / support, come over to
>>   https://plus.google.com/communities/115879488566665599508
>> ***************************************************************
>>
>> WARN:Connecting to Port /dev/ttyAMA0,115200
>> WARN:Connected to /dev/ttyAMA0 at 115200
>> Recv: error:2
>> Recv: error:2
>>
>>
>>
>>
>>
>>
>>
>>> server.js shows its side on the console. The two chips have incompatible
>>> IO voltages. Are you using a 3.3v - 5v level shifter?
>>>
>>> <https://github.com/gnea/grbl/issues/100#issuecomment-274283939>, or mute
>>> the thread
>>>
>>
>>
>>
>> Kind regards,
>>
>> *Bertus Kruger*
>> *Mobile - 021 078 5392*
>>
>>
>
>
> Kind regards,
>
> *Bertus Kruger*
> *Mobile - 021 078 5392*

closed 2017-01-21 by Protoneer