gnea/grbl #98: How can my GUI determine how big the serial buffer is on a GRBL implementation?

cspwcspw, 2017-01-21:
Grbl 0.9 reported the number of bytes in use in the serial buffer,  0 meant the buffer was empty.  In V1.1 the report changes to "number of serial bytes free".   That makes some tasks of the sender easier.

I'm looking at this issue  [https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl]
  "_When Grbl stores data to EEPROM, the AVR requires all interrupts to be disabled during this write process, including the serial RX ISR. This means that if a g-code or Grbl $ command writes to EEPROM, the data sent during the write may be lost. This is usually rare and typically occurs when streaming a G10 command inappropriately inside a program. For robustness, GUIs should track and detect these EEPROM write commands and handle them appropriately **_by waiting for the queue to finish executing before sending more data_**._"

So I'd like to know if the serial queue is empty.  Previously that was easy.  Is there a way to determine how big the serial buffer is on an implementation so that I can tell when it is empty?  (Indirectly, I guess forcing a processor reset followed by a status query will shows available capacity.  I don't really want to go that route. )  Are there other suggestions or is there a simple way to detect the buffer capacity?

Thanks

swarfer, 2017-01-21:
Hi Pete
The moment your GUI connects the Arduino will reset, so your first status request will tell you the buffer size, I think.

However, one really should not send EEPROM store commands in the middle of a job....

cspwcspw, 2017-01-21:
Thanks Dave.  

I'd prefer not to trust the user to be well-behaved (and I also want to misbehave), so I have some reasons why I want to stream EEPROM commands.   

I already have a mechanism to wait for an empty buffer in my V0.9 sender, so its just my update to V1.1 that triggered my post.   I think easiest workaround is just to remember the high-tide of any free bytes I've seen in a status report, and assume that when I see that again it means "empty buffer".

My use-case for wanting to stream EEPROM writes (or reads) is that I have a machine in the garage, and another Arduino on my desk (without shield or motors) where I can dry-run jobs and test code generation. (But I also use the desk Arduino for some non-grbl projects, so it often gets overwritten.)  So my GUI wants to be able to fetch or install settings to a grbl.  And with settings like XYZ limits, I find myself having to get them consistent in 3 places: the garage grbl, the desk grbl, and in the GUI, so that canvas or 3D space in the GUI exactly matches the machine's notions of its limits.  So being able to automatically interrogate them or stream them to grbl sounds appealing, even if they don't play so nice with EEPROM ...

Do you know if the Mega also has this same interrupt lock-out when accessing its EEPROM?
 
Thanks
Pete

swarfer, 2017-01-21:
Alright, so long as you know what you are doing (-:

I believe the Mega has the same limits since it is in the same uP family.

swarfer, 2017-01-21:
PS: I must come and see this machine sometime!

electrokean, 2017-01-21:
Interrupts aren't disabled during EEPROM read, only during write.
I don't have an Arduino here to test, but I'd assume that when interrupts are disabled during the EEPROM write, you won't get any responses to the status command at all.  And pretty sure you also only get the ok response after the write completes.
The big problem is that while the interrupts are disabled, serial characters are likely to be lost as the receive buffer can't be emptied.  So if you know which commands cause EEPROM writes, you just need to avoid further streaming until you get the "ok" and/or start getting status responses again.

chamnit, 2017-01-21:
Unfortunately, the size of the serial buffers and the planner buffer was something I forgot to add to v1.1. I intended to put it into the $I build info string so GUI can query Grbl for that data. It'll be in v1.2 when I get to it, but I'm busy with porting to ARM. 

You can stream Grbl settings and safely write them if you use the send-response protocol. It's safe because it doesn't send more serial RX data after you send a save command. You can test with the simple_stream.py script. It works. I do it all of the time.

closed 2017-01-21 by chamnit

chamnit, 2017-02-01:
@cspwcspw @cprezzi : FYI. Today's release of Grbl v1.1f includes total planner blocks and serial RX bytes in the `$I` build info string. See release notes for details.

cprezzi, 2017-02-02:
@chamnit Thank you for the info Sonny!

cspwcspw, 2017-02-03:
Appreciate this!  Thanks again for the huge effort.