gnea/grbl #101: Configuration GRBL 1.1

robcop2000, 2017-01-22:
After the upgrade to GRBL 1.1e  I had some problem.
First the old electronic make strange behaviors, so I have changed everyting, Arduino, Stepper motors, CNC shield etc...All new.
Now the configuration has strange behaviors.
1 - The $3 – Direction port invert, mask doen't change any value I put inside.
2- I have the same motor, driver on the CNC shield both are set to 1/8 for the X and Y axis. Why I have to insert in $100, $101  [X,Y] steps/mm two different value to have the correct movement?
3- Now the CNC controller saies that the GRBL is no connected, but if I made jog movements it works?

109JB, 2017-01-22:
You need a different or updated control software that works with v1.1

Things were changed in how controllers interface and a program that works with v0.9 will not work with v1.1 unless the author has changed it

robcop2000, 2017-01-22:
I'm using the last version of bCNC the If I read correct should be one that works

chamnit, 2017-01-22:
@robcop2000 : Sounds like you might some kind of electrical or wiring issue. What happens when you disconnect everything except a bare Arduino to bCNC?

Also, which version of Grbl are you upgrading from? v0.9? v0.8?

robcop2000, 2017-01-22:
The grbl was 0.9j worked fine

chamnit, 2017-01-22:
@robcop2000 : Did you use v0.9j's default build? Or did you disable something like the variable spindle output?

And how are you uploading Grbl v1.1? Did you remove all previous installations of Grbl from your computer's Arduino library?

robcop2000, 2017-01-22:
I have used default. 

I have uploaded the .hex file

chamnit, 2017-01-22:
@robcop2000 : Did you try disconnecting everything from the Arduino? If not, please do that. It doesn't matter if v0.9 worked before. 

Second, try uploading via the Arduino IDE method. See the Wiki on how to do that. It's fairly easy.

robcop2000, 2017-01-22:
Yes to flash the Arduino I have to disassemble the board. The computer is far from the CNC

I will try with the IDE method, It will take some time.

robcop2000, 2017-01-22:
I have taken another Arduino Board, Flashed with IDE arduino 1.8.1 but the problem still exist

vMeph, 2017-01-22:
can you place here your grbl settings $$
i find weird you saied
> Why I have to insert in $100, $101 [X,Y] steps/mm two different value to have the correct movement?

you have to use the formula to put the correct values for your motors steps
do you have saved the settings you add  set on the version grbl  0.9 that you add before?
cause when upgraded for the new version 1,1 your settings was erased and you got to configure your machine

robcop2000, 2017-01-23:
This is my GRBL  config
GRBL 1.1e Config

$0=10
$1=25
$2=0
$3=4
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=80.000
$101=40.000
$102=400.000
$110=2000.000
$111=2000.000
$112=200.000
$120=500.000
$121=500.000
$122=50.000
$130=830.000
$131=1830.000
$132=170.000

I know this doesn't make sense to me too
Unfortunately I didn't save the old GRBL config.

robcop2000, 2017-01-23:
Now the situation is this.
I Have flashed the Arduino board on the PC and configurated.
I have changed the USB cable ( Was the only thing old)
I have installed on the machine an seemes to work.

The only problem is Homing, that first the Z axis runs in Home position, after stops when  Spindle reach the X or Y limit, and not both like before.

chamnit, 2017-01-23:
@robcop2000 : Ok, so it was the USB cable. With your homing issue, it's not clear what the problem is.

robcop2000, 2017-01-23:
Before the update when you send home the machine:
1. Run home position with Z axis
2. Run both X and Y axis to home position. 
3. When X or Y reach the home position the other axis continue to run until it reach home.

Now with after update GRBL 1.1e the machine
1. Run home position with Z axis
2. Run both X and Y axis to home position. 
3. When X or Y reach the home position the machine stops there. So X or Y isn't in Home position.
It depends which axis has a shorter distance to run

chamnit, 2017-01-23:
@robcop2000 : I would check your wiring. It sounds like your X and Y limits are crossing into each other somehow.

robcop2000, 2017-01-23:
This seemes to me strange. If I click one limit at a time they all works. 
For checking I have to disassemble the machine. I will do it later.

There isn't the way to understand which is the limit that has gone in alarm?
Should be good feature

chamnit, 2017-01-23:
@robcop2000 : There is no way to know which switch caused an alarm with 100% certainty, because the AVR pin change interrupt doesn't track which pin causes it to fire. You could poll, but it can provide a false reading. Closing this issue since this is now resolved in terms of Grbl.

closed 2017-01-23 by chamnit

robcop2000, 2017-01-23:
I don't know if is a wiring problem I have to check

robcop2000, 2017-01-24:
I check all the wiring of the limit switch but the problem is still present.
I'm reading in the wiki.
"USB to Serial transmission errors"
I have an genuine Arduino Uno with  Atmel 16U2 chip.
can be this the problem?

chamnit, 2017-01-24:
@robcop2000 : If it's communicating and working with the new USB cable, then it's not USB. Homing operates without USB communication until the cycle is finished. You'll be able to check which pins Grbl is reading as triggered in its '?' status reports in the `Pn:` field. If it doesn't show up, that means that Grbl doesn't detect any input pins as triggered. If you trigger and hold down the x-limit, it'll show an `Pn:X`.

robcop2000, 2017-01-25:
I have tryed to see the report but there is no Pn: Value
Below the commands
Thanks
in advance
Roberto

Grbl 1.1e ['$' for help]
$X
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
ok
$$
$0=10
$1=25
$2=0
$3=4
$4=0
$5=0
$6=0
$10=31
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=80.000
$101=40.000
$102=400.000
$110=2000.000
$111=2000.000
$112=200.000
$120=500.000
$121=500.000
$122=50.000
$130=830.000
$131=1830.000
$132=170.000
ok

$H
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Idle|MPos:-829.000,-1829.000,-1.000|Bf:15,124|FS:0,0>
?
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok

ALARM:1   (I PUSH THE X SWITCH LIMIT)
$X
[MSG:Reset to continue]
Grbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
ok
<Idle|MPos:-829.000,-1829.000,-1.000|Bf:15,127|FS:0,0>
?
ok

ALARM:1    (I PUSH THE Y SWITCH LIMIT)
$X
[MSG:Reset to continue]
Grbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
ok
<Idle|MPos:-829.000,-1829.000,-1.000|Bf:15,127|FS:0,0>
?
ok

ALARM:1   (I PUSH THE Y SWITCH LIMIT)
$X
[MSG:Reset to continue]
Grbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
ok
<Idle|MPos:-829.000,-1829.000,-1.000|Bf:15,127|FS:0,0>
?
ok

vMeph, 2017-01-25:
Just a question is your motors all diferent from eatch other?
Cause your settings $100,$101,$102 looks a litle weird

robcop2000, 2017-01-25:
Yes 
The machine has a working are of 830 mm x 1830 mm x 170 mm
I'm an architect and I use it to build furniture prototype and architectural models.

Motors type
Z= Nema 17 0,9° with screw (Z axis doesn't need high torque)
X= Nema 23 0,9° with belt GT2 (X axis was the original)
Y= Nema 23 1,8° with belt GT1 (Y axis has been changed becaus it was equal to X motor an this is the one I found here. Another think in reason of the Y axis is moved by a single motor I need more torque).

robcop2000, 2017-01-26:
Chamnit
you wrote:
...
_ou'll be able to check which pins Grbl is reading as triggered in its '?' status reports in the Pn: field. If it doesn't show up, that means that Grbl doesn't detect any input pins as triggered. If you trigger and hold down the x-limit, it'll show an Pn:X._
...
As you can see in the report a day ago GRBL doesn't show up any input as triggered, but If I hold down any switch limit GRBL shows correctly ALARM

At the same time the behaviour you assume was a
.......
_It sounds like your X and Y limits are crossing into each other somehow._
.......
I have tryed to disable the Y switch limit (I remove cables from the screw connector) so in my mind the Homeing should first go to Z home, after X Home, and continue to Y homeing.
In reality, first runs to Z homeing and after at X homeing stops.

Any advise?
Thanks in advance

vMeph, 2017-01-26:
Did you try $27 the homming pull off to give lets say 5mm maybe when z goes stays on top of switch with only 1mm?

robcop2000, 2017-01-26:
Yes I have tryed now but no difference

chamnit, 2017-01-26:
@robcop2000 : It's really not clear what's going on, because when stated that homing works, except that both XY axes stop when it triggers either one of them. That means your limits are being read by Grbl. Grbl will show the _current_ limit state whenever you query it, but that also means that it must be engaged while you ask it for a report.

robcop2000, 2017-01-26:
So you mean that I have to wait a while before to ask with ?

robcop2000, 2017-01-26:
to ask the report

chamnit, 2017-01-26:
@robcop2000 : The switch needs to be actively triggered while you query it. The easiest way is to move your machine to trigger one and keep it triggered, then query it. Or take your Arduino out and just connect a limit pin to ground with a wire.

robcop2000, 2017-01-26:
Chamnit
These are the respone doing as you said. I trigger one axis a time.

**X axis Triggered**
ALARM:1
?
[MSG:Reset to continue]
Grbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
<Alarm|MPos:-93.138,0.000,0.000|Bf:15,127|FS:0,0>
ok

**Z Axis Triggered**
ALARM:1
?
[MSG:Reset to continue]
Grbl 1.1e ['$' for help]
[MSG:Check Limits]
[MSG:'$H'|'$X' to unlock]
<Alarm|MPos:-83.138,0.000,16.115|Bf:15,127|FS:0,0|Pn:PXYZ>
ok

**Y axis Triggered**
ALARM:1
$X
[MSG:Reset to continue]
Grbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
ok

gerritv, 2017-01-26:
You can also use a GUI that latches any P report value. e.g. GrblPanel remembers seeing a limit trigger from responses to regular ? commands. You have to manually clear the indication on the GUI. You are going to find it difficult using just a serial monitor IMO.

Your info above indicates that all the pins were triggered, Probe, X, Y and Z when you trigger the Z limit. I think you have a wiring problem.

kenansatir, 2017-11-02:
Please use RFI, EMI, EMCi Noise Filters on limit switch cables (near to board) like :

https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20171102041903&SearchText=+RFI+EMI+EMC+Noise+Filter

But try to use big ones... about 2cm diameter....

Also connect your boards GND pin to ground...

Best regards

Andre1965ru, 2021-12-21:
нет ответа GC:G0 G54 G17 G21 G90 G 94 M5 M9 T0 F0 S0

Andre1965ru, 2021-12-21:
how to configure CNC xPRO V5