gnea/grbl #105: Using external hardware to boost the GRBL

luben111, 2017-01-25:
Hi,

I have seen many CNC controllers and many of them have dedicated FPGA to generate the STEP/DIR signals. For example on my CNC Sieg KX1 (http://www.arceurotrade.co.uk/Catalogue/Machines-Accessories/CNC-Milling-Machines/Sieg-KX1-CNC-Hobby-Mill/Sieg-KX1-Hobby-CNC-Mill-Bundle) is used LeafBoy controller which is capable to generate 200KHz pulses  
http://leafboy77.com/index.php/en/
[image]

Do you think it's worth designing such FPGA to improve the performance of GRBL? The FPGA offers ultimate hardware features - we can add any kind of timers and pulse generators inside.  I have a lot of experience in designing FPGA so this should be not that difficult.

chamnit, 2017-01-25:
@luben111 : Yes. Dedicated FPGA chips are often on industrial machines serving as step/pulse generators for this very same reason. However, the hobby electronic market doesn't really have any FPGA tools and hardware that is cheap, available, or well-supported by the community. The point of Grbl is to make CNC widely accessible, which is why its crammed into an Arduino. 

If you haven't been following the ARM discussion for the next step of Grbl, the next project will be ported to several ARM chips. My early tests on a SAMD21 have been showing nearly 100kHz step rates without USB CDC enabled (comes down to 60kHz with it enabled). Other users have ported Grbl to LPC, which has shown to support over 200kHz easily. Even faster ARM chips would push this higher. There is also the question how high of a step pulse frequency you actually need. For hobby applications, 100kHz will do the job for 99% of applications for most people. You likely don't need anything above 250Khz realistically and ARM chips can do it.

closed 2017-01-25 by chamnit

luben111, 2017-01-25:
I totally agree - GRBL should stay as simple as possible, that's why I really like it :)

tdkpipo, 2017-04-15:
compañero, si realizaste el proyecton fpga 
necesito el informacion al respecto para un proyecto en mi universidad