gnea/grbl #108: M3 Works, M4 Does NOT Work. (Video To Issue Linked)

inciteman, 2017-01-27:
Hello, so I'm having an issue with the **M4** Command on an Arduino Uno with GRBL 1.1e. I understand it ONLY works when the G1 command is used. The command **M3 S50** works fine, pin D13 the "Spindle Direction" goes LOW (enabling the laser) and D11 the "Variable Spindle PWM" modulates. However, when I send **M4 S50** and then follow it with a **G1 X10 F100** then the pin D13 stays HIGH (disabling the laser) but the D11 DOES modulate.

Link to Video: https://www.youtube.com/watch?v=uei6Dhy-Acw

I'm surprised I'm the only one with this issue. Maybe it's a hardware problem? Operator? Murphy's Law?

Any help would be greatly appreciated.

chamnit, 2017-01-27:
@inciteman. Please read the wiki page on the laser mode works.

closed 2017-01-27 by chamnit

inciteman, 2017-01-27:
Hello chamnit, thank you for the reply! However, I've read the "Grbl v1.1 Laser Mode" page about 7 or 8 times now for the past 2 days.

I'm fully aware that in the M4 laser mode, the laser ONLY works when in G1, G2, or G3 motions and mine does PWM (Pin D11) properly; however, it does NOT set the D13 LOW (so the CO2 laser will NOT turn on).

M3 on the other hand works fine, D11 will PWM and D13 gets set to LOW (turning on the laser).
M4 will PWM D11 but does NOT set D13 LOW, it stays HIGH (leaving the laser OFF).

Sorry for the repeated question.

chamnit, 2017-01-27:
@inciteman : The spindle direction pin D13 is a direction pin, not an on/off enable pin. M3 and M4 are CW and CCW in g-code, so M4 will turn on D13 when enabled. If you need an on/off, you need to edit the config.h file and enable the USE_SPINDLE_DIR_AS_ENABLE (or something similar looking) option

inciteman, 2017-01-27:
Okay now I understand the concept a bit better. So normally it's used for milling, but in Laser mode ($32=1)  it does not change its behavior (because normally laser DIODE's only require a PWM to fire). 

So I enabled USE_SPINDLE_DIR_AS_ENABLE but like the description states _"With no direction pin, the spindle clockwise M4 g-code command will be removed. M3 and M5 still work."_ and now M4 under the laser mode does not work at all, and M3 goes HIGH now instead of low (I'm sure it can be inverted somehow) but unfortunately this disables the M4 laser mode which I don't want to do.

chamnit, 2017-01-27:
@inciteman : Ah, you're right about that. In laser mode, M4 is different. Preventing M4 from operating seems a little heavy handed when USE_SPINDLE_DIR_AS_ENABLE is enabled. I'll think about removing that stipulation so that it works with laser mode. For now, I think all you have to do is remove the #ifdefs on lines 255 and 262 of gcode.c to get it operational. Let me know how it works.

inciteman, 2017-01-27:
Yes, that would be great!

I removed 255 and 262 (w/it's contents), and it disabled the M4 command. Now M4 returns _error:20_

         // #ifndef USE_SPINDLE_DIR_AS_ENABLE_PIN
         //   case 4:
         // #endif
          case 3: case 5:
            word_bit = MODAL_GROUP_M7;
            switch(int_value) {
              case 3: gc_block.modal.spindle = SPINDLE_ENABLE_CW; break;
         //     #ifndef USE_SPINDLE_DIR_AS_ENABLE_PIN
          //      case 4: gc_block.modal.spindle = SPINDLE_ENABLE_CCW; break;
          //    #endif

chamnit, 2017-01-27:
@inciteman : Sorry I meant the defines should be removed but what is inside them shouldn't be. So both case 4: statements need to be uncommented in your example.

inciteman, 2017-01-28:
@chamnit 
Oh ok, tried that. Still get the same behavior (M3 Works fine, M4 PWM works but D13 (Direction pin) stays HIGH)

        //  #ifndef USE_SPINDLE_DIR_AS_ENABLE_PIN
            case 4:
        //  #endif
          case 3: case 5:
            word_bit = MODAL_GROUP_M7;
            switch(int_value) {
              case 3: gc_block.modal.spindle = SPINDLE_ENABLE_CW; break;
        //      #ifndef USE_SPINDLE_DIR_AS_ENABLE_PIN
                case 4: gc_block.modal.spindle = SPINDLE_ENABLE_CCW; break;
         //     #endif

Just did a work around for now.  Should NOT be used when milling. In spindle_control.c replaced lines 119 to 123 with 

    if (pwm_value == SPINDLE_PWM_OFF_VALUE) {
      SPINDLE_TCCRA_REGISTER &= ~(1<<SPINDLE_COMB_BIT); // Disable PWM. Output voltage is zero.
	  SPINDLE_DIRECTION_PORT |= (1<<SPINDLE_DIRECTION_BIT); // Set pin to HIGH when PWM is disabled (LASER WORKAROUND)
    } else {
      SPINDLE_TCCRA_REGISTER |= (1<<SPINDLE_COMB_BIT); // Ensure PWM output is enabled.
	  SPINDLE_DIRECTION_PORT &= ~(1<<SPINDLE_DIRECTION_BIT); // Set pin to LOW when PWM is enabled (LASER WORKAROUND)
    }

Not the best way to code it for release but it works exactly as expected with M3 and M4 for now.
Thank you for the help!

chamnit, 2017-01-28:
@inciteman : Just as a clarification, the enable pin is different than the PWM pin. It will stay enabled as long as the M3 or M4 is still active. It doesn't toggle when spindle speed is zero.

chamnit, 2017-01-28:
@inciteman : I knew this looked familiar. I had just fixed this issue, two commits ago, where Grbl was setting the enable pin with S0 and S>0. A user asked make the enable pin independent of S, because he often enables his laser with M3/4 prior to starting a job as part of his work flow. Seeing that there are multiple workflows, I'll add a compile-time option to alter the behavior of enable pin, based on what you want.

jgilbertfpv, 2018-12-13:
@chamnit I'm having a similar issue, but I'm guessing there's been some updates since this post and I don't see some of those pieces of the code when looking for it. I'm using M3 fine, but M4 doesn't turn on the laser. I'm on the arduino mega instead of the 328p. I have my laser PWM connected to pin 7 Spindle PWM and laser fire is conneted to pin 5 Spindle Direction and I have enabled laser mode. I've read over the laser mode settings and tried several things and google led me here. Am I doing something wrong?