gnea/grbl #109: Grbl controller not showing the position

adibiut, 2017-01-27:
Hi. I use Grbl controller as the GUI. It was ok with v0.9 but in v1.1 it doesn't show the Machine and work coordinates (it's all zero). Why?

luben111, 2017-01-27:
I'm using v1.1 and I can see machine coordinates dynamically changing while moving. GRBL v.1.1 has some minor changes compared to previous 0.9xx - it can't show both machine and working coordinates, also some additional information is appearing in the report which can fool the PC program

gerritv, 2017-01-27:
GrblController was not updated to the new Grbl 1.x protocol. Zapmaker seems to have disappeared according to some reports.

closed 2017-01-27 by chamnit

adibiut, 2017-01-27:
Thanks.
Which GUI do you suggest to work with v1.x?

f0m3, 2017-01-27:
I worked with Universal Gcode Sender and also found a new one: [https://github.com/cncjs/cncjs/wiki](https://github.com/cncjs/cncjs/wiki) that one looks promising to me.