gnea/grbl #111: Homing cycle problems

ThomasDalgas, 2017-01-28:
Hi 

I'm unable to getting Homing activated via endstops..
I'm using the Protoneer Raspberry Pi Board and grbl 1.1e firmware installed.
I have asked the Protoneer forum, but was advised to ask here, as it could be a grbl issue.

I have $21 Hard limits and $22 Homing cycle enabled, and when pressing endstops i get an error 9
and i have to unlock board to continue.. 

When i activate Homing Cycle, nothing really happens..

This is done via bCNC and CNC 1.8.. 
Is there something in the hat i have missed?

Everything else is working fine.

f0m3, 2017-01-28:
Can u post your configuration and how u wired the limit switches?

luben111, 2017-01-28:
I use the save V1.1e and the homing works perfectly well. If possible post the configuration settings

chamnit, 2017-01-28:
Make sure the Z-limit pin is on D12, not D11.

Protoneer, 2017-01-28:
The Z-lim pin is correct on the board as it is working okay with the last
v0.9

chamnit, 2017-01-28:
@Protoneer : Thanks for the clarification. There isn't much different in v1.1's homing cycle or limits handling, if at all. Now that I re-read this, it's odd that an error:9 comes up when pressing a limit. This comes up when a g-code command is sent to Grbl during an alarm state. In this case, I think it would be best to isolate the issue by removing the Arduino completely and test it isolated with Grbl only. 

Also @ThomasDalgas  please share your Grbl $$ settings, $I build info, and any other config.h changes you have made. Thanks.

ThomasDalgas, 2017-01-28:
My grbl settings:
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=100.000
$25=500.000
$26=250
$27=6.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=3000.000
$131=3000.000
$132=100.000
ok

It's connected with the end stop XYZ with common ground..
[http://wiki.protoneer.co.nz/images/1/13/RPI-CNC-V258-CAD-Front.jpg](url)

My endstops are Makerbot 
[http://beta.ivc.no/wiki/images/7/73/MakerBot_end-stop_schematic.png](url)
[http://beta.ivc.no/wiki/images/7/79/MakerBot_end-stop.jpg](url)

Right now i'm testing with a single NO switch (pushbutton) directly on the inputs, with the same results..

chamnit, 2017-01-28:
@ThomasDalgas : Nothing super unusual so far. What does your $I build info string print out?

luben111, 2017-01-28:
Do you have really 3m x 3m movement on XY?
Your homing speed is 500mm/min ($25) , it should take up to 6 minutes to complete the homing. I would not be surprised if there is some timeout for GRBL to complete the homing.

ThomasDalgas, 2017-01-29:
No, i had just wrote some numbers in and forgot about that, as i was focused on this Homing issue.
But still no luck, i have totally wiped the EEProm, and uploaded 1.1e again, just to try something.

To give an idea what is happening i have copied the text from UGS in verbose mode here.. :

Grbl 1.1e ['$' for help]
>>> $X
ok
>>> $H        // After $H is send, the verbose stops totally and nothing happens, no timeout. When i                                        //                     press endstop on Z axis, it getting this alarm after 15 secs.

ALARM:9
ok

Grbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0|WCO:0.000,0.000,0.000>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0|Ov:100,100,100>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0>
[verbose]<Alarm|MPos:-10.000,-10.000,5.000|FS:0,0|WCO

It's getting more and more strange for me..

luben111, 2017-01-29:
There is something wrong with Z axis - it should be negative! Check is homing directions were set correctly.

chamnit, 2017-01-29:
@ThomasDalgas : ALARM:9 is totally different from error:9. ALARM:9 indicates "Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring." As shown in the documentation. 

Also, homing is blocking and will not show any feedback until it completes.

Can you explain in detail what the homing cycle does. Step by step. This will help.

@luben111 : Position is fine. Grbl boots at 0,0,0 and it is searching in the positive direction for Z.

chamnit, 2017-01-29:
@ThomasDalgas : Did you update your step/mm? They are still 250 steps/mm. Are all of your settings exactly the same as they were in v0.9? It doesn't seem so.

ThomasDalgas, 2017-01-29:
I will try some of the these tomorrow. This is my first CNC build, so i'm newbie to all this. The CNC  has never been running with the 0.9. But i'm getting more and more confidence with grbl, so i'm learning out of this trial and error.. I will report back what i figure out here..

ThomasDalgas, 2017-01-29:
and i appreciate the fast response i'm getting here.. Thanks alot.. :-)

Flyingbobcats, 2017-01-30:
New to grbl, but I'm having a somewhat similar issue. The homing / limit switches are NO and wired to 9,10, and 12 respectively.  The other end of the switches are straight to ground. With $21 and $22 enabled, I command the grbl to home. The z will home correctly (about half of the time) and the x and y axes will then move in a jerky/jittery motion for about 1/4" travel before stopping. Attempting the home several times and would get different behavior. The z would move in a jerky motion, followed by y up until it hit the limit switch and kept going. No errors or warnings displayed.  I'm wondering if there is an issue with my wiring or possibly something in firmware. 
 
Settings:

Grbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
$0=250
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=50.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=100.500
$101=100.500
$102=100.500
$110=2000.000
$111=2000.000
$112=2000.000
$120=1000.000
$121=1000.000
$122=1000.000
$130=200.000
$131=200.000
$132=200.000
ok
ok

luben111, 2017-01-30:
@Flyingbobcats 
>> and the x and y axes will then move in a jerky/jittery motion for about 1/4" travel before stopping

If you experience jerky/jittering movement you probably need to adjust acceleration and max speed for these axis - try to decrease them.

Other possibility is that stepper motor current is very low and need to be increased.

Flyingbobcats, 2017-01-30:
@luben111, thanks for the quick reply. The acceleration was giving me an issue a few days ago when I started setting up the system. Adjusted it and it will move fine at low, medium, and high speeds. Tried homing again a few minutes ago, and still had the same problem. Sometimes it will behave like a limit switch was reached when it hasn't...other times not recognizing a limit was thrown at all and keeps moving. I'm wondering if I need a pull up resistor somewhere. Need to read the guides again

luben111, 2017-01-30:
@Flyingbobcats 
If you have long wires for homing switches and all they pass in vicinity to stepper motor wires (where large currents are flowing) you may get wrong homing switch detections. You can add extra pull ups to 5V (1K to 4.7K) which may improve things. I would recommend you add optocouplers (see thread https://github.com/gnea/grbl/issues/96 )

But if you have jerky/jittering motion I'm more inclined that you have problems with stepper motors or mechanic issue:
- the current of  steppers is too low
- the feed rate and accelerations are too high
- you have some bad junctions in the wiring of steppers   
- there is some mechanic issue (higher friction in some particular areas)

Jerky/jittering movement has nothing with homing and is good indicator of some other problems.

electrokean, 2017-01-30:
@Flyingbobcats 
I agree with @luben111 that what you're describing is very likely noise on the limit inputs.
The "jerky/jittering motion" could actually be due to the homing pull-off also seeing noise - hard to tell without seeing it.
Read #96 and try adding 1k pull-ups and 100nF capacitors on the limit inputs.  If you don't have any readily available, you could try adjusting $26 for de-bouncing in software.

Flyingbobcats, 2017-01-30:
@luben111 @electrokean, thanks for the discussion.

@electrokean, what you're describing sounds accurate, I was simply unable to describe the movement. The jerky movement does look like a homing pull-off. Other issues could be with the setup I'm currently operating at. Instead of installing headers, I soldered the wires straight to the board and had one heck of a time. Going to install headers to make troubleshooting a little easier. I'll add in the caps and pull-up resistors and see if that helps. Thanks guys, I'll let you know

chamnit, 2017-01-30:
@Flyingbobcats : I would also reduce your `$0=250` setting to something like `$0=10`. You won't be able to go over 3500 steps/sec, if you don't, because the step pulses will take longer to end than the time until the next one comes.

ThomasDalgas, 2017-01-31:
So, got some time off to visit my little project, and i actually got it to work. Yes it was all about config.. found this on another forum for the MPCNC (My build), and some guy posted this config..

$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=200.000 (x, step/mm)
$101=200.000 (y, step/mm)
$102=4535.440 (z, step/mm)
$110=1800.000 (x max rate, mm/min)
$111=1800.000 (y max rate, mm/min)
$112=180.000 (z max rate, mm/min)
$120=100.000 (x accel, mm/sec^2)
$121=100.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=960.000 (x max travel, mm)
$131=550.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

After i took the details i needed, everything just worked right away. 
So for now i just need to adjust the CNC and calibrate the hell out of it..

Thanks everybody for the input.

closed 2017-01-31 by ThomasDalgas