gnea/grbl #112: USE_SPINDLE_DIR_AS_ENABLE_PIN

poesel, 2017-01-28:
Hi,

I have both
#define VARIABLE_SPINDLE
and 
#define USE_SPINDLE_DIR_AS_ENABLE_PIN
enabled.

When measuring from D11 to ground the voltage is as expected. But measuring between D13 and ground always yields 0V.

Now I'm a bit at a loss how to fix this. Thanks!

chamnit, 2017-01-28:
@poesel : Are you commanding an M3 or M4?

poesel, 2017-01-28:
This is the program I use for testing:
%
(1001)
G90 G94
G17
G21

(Schlitze)
M9
T1 M6
S12000 M3
G54
M9
G1 X0 Y0 Z0 F100
G1 X50 Y50 Z50
G1 X0 Y0 Z0
(G1 Z3 F1000)
M30
%

chamnit, 2017-01-28:
@poesel : Ok. Grbl $$ settings, $I build info, and anything else you have changes would help.

poesel, 2017-01-28:
$I
[MSG:Check Limits]
[MSG:'$H'|'$X' to unlock]
[VER:1.1e.20170114:]
[OPT:V]

$$
$0=40
$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=0
$21=1
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=24000
$31=1
$32=0
$100=79.918
$101=79.918
$102=530.000
$110=2000.000
$111=2000.000
$112=500.000
$120=50.000
$121=50.000
$122=10.000
$130=1400.000
$131=600.000
$132=75.000

poesel, 2017-01-28:
[config.h.txt](https://github.com/gnea/grbl/files/736944/config.h.txt)

File renamed to be able to upload...

chamnit, 2017-01-28:
@poesel : Does pin D13 go high when you command an M4? If so, you are not editing the correct config.h file. It should be one in the Arduino library folder not the one you downloaded.

poesel, 2017-01-28:
*facepalm* aaargh - I'm an idiot! :)

Of course I edited the wrong file. Build options now shows 'VN' (is there a list somewhere?) and now I get the 5v on D13.

Thanks and sorry!

closed 2017-01-28 by poesel