gnea/grbl #113: Homing Status updates

msraynsford, 2017-01-29:
I have a program that issues a homing command to grbl and waits for it to finish homing. 
In grbl 0.9 I would poll the current status command '?' and wait while while the status was 'HOME'

grbl 1.1 doesn't seem to reply to the current status command while it is homing and so I end up with a serial timeout or an indefinite blocking delay.

Any ideas on how to fix this or work around this?

chamnit, 2017-01-29:
@msraynsford : You sure about v0.9 reporting during a homing cycle? The homing cycle isn't at all much different in v1.1 from v0.9. The only thing I could think of is that it may report once or twice, where v1.1 may not.

msraynsford, 2017-01-29:
Wow super fast response. It was reporting enough to not cause a timeout on my serial, guess I was just lucky.
I currently have a 'how long is a piece of string' issue and I'm not sure how best to solve it. There is no response from GRBL at the end of a cycle so I don't know when to stop waiting.

Alternately I'd be happy for grbl to auto home when it starts up but I'm unsure how to make it do that  as the startup blocks are executed after a homing cycle occurs.

chamnit, 2017-01-29:
@msraynsford : Auto-homing is dangerous, which is why Grbl doesn't support that type of option. FWIW, allowing status reports during a homing cycle was supposed to be added to v1.1, but did not make it in. In future versions of Grbl, I'll fix this, but for now, it will have to wait.

closed 2017-01-29 by chamnit