gnea/grbl #115: G10 Offset Persistence

scottmil, 2017-01-31:
I am a new GRBL user trying to use the G10 command to reset the WCO X0 Y0 origin from the top right corner (homing position) of my laser engraver to the bottom left corner using GRBL version 1.1e.20161208.  After successfully homing, I manually jogged my carriage to the lower left corner, then issued the G10 L2 P1 X0. Y0. command, but the I cannot get the new WCO X0 Y0 offset to persist.  Here is my sequence of commands with ***comments:

***Laser Mode ($32=1) in settings
$H
ok
***carriage homes correctly at top left corner***

?
<Idle|MPos:-2.996,-2.996,0.000|Bf:15,126|FS:0,0|WCO:0.000,0.000,0.000>
ok
*** home mPos and WCO coordinates

G91 X-198. Y-225. F500
ok
***move carriage to lower left corner

<Idle|MPos:-200.992,-227.996,0.000|Bf:15,126|FS:0,0|Ov:100,100,100>
ok
***lower left corner MPos coordinates

G10 L2 P1 X0. Y0.
ok
***set lower left corner for WCO X0 Y0

?
<Idle|MPos:-200.992,-227.996,0.000|Bf:15,126|FS:0,0|WCO:0.000,0.000,0.000>
ok
***confirm that lower left corner is new WCO X0 Y0

?
<Idle|MPos:-200.992,-227.996,0.000|Bf:15,126|FS:0,0|Ov:100,100,100>
ok
***subsequent ? does not display WCO

$#
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
***shouldn’t G54 persist G10 offsets here? Instead shows all zeros

G0 X50. Y50. F500
ok
*** carriage correctly moves to X50 Y50 using new WCO origin based on G10 setting

G0 X0. Y0. F500
***return to G10 origin fails, carriage does not respond to command, no carriage movement

?
<Idle|MPos:-150.992,-177.996,0.000|Bf:15,126|FS:0,0|WCO:0.000,0.000,0.000>
ok
***current carriage MPos  

———————————————————————————————————
Notes:
1. Disabling laser mode $32=0 has same result as above.  G10 setting does not appear to be persisted.
2. Only the first G0 movement after setting WCO to lower left corner with G10 appears to succeed.  Second G0  command after setting new WCO origin with G10 results in no movement or incorrect carriage movement.
3. All commands issued with Arduino 1.6.12 IDE Serial Monitor. 
4. GRBL settings and config.h attached.

[grbl_settings.txt](https://github.com/gnea/grbl/files/743435/grbl_settings.txt)

[config.h.txt](https://github.com/gnea/grbl/files/743437/config.h.txt)

scottmil, 2017-01-31:
Oops.."carriage homes correctly at top left corner" should read "carriage homes correctly at top right corner".

chamnit, 2017-01-31:
@scottmil : Everything is working fine. Please read how WCO works in the wiki. It is intermittent to reduce communication overhead. Second, please read how G10 works in the LinuxCNC g-code descriptions. You are setting a [0,0,0] work coordinate offset with a `G10 L2 P1 X0 Y0 Z0` command. You probably meant to use `L20` rather than `L2`.

closed 2017-01-31 by chamnit

scottmil, 2017-02-01:
Thanks!  G10 L20 P1 X0 Y0 Z0 solved my problem. I had initially tried G10 L20 instead of G10 L2 but was confused when I interpreted the results of the subsequent ? command which I thought should have reported 0.000,0.000,0.000 for the WCO origin rather than the offset from the machine home position.  (Hope I got this response right.)  $# is also now reporting the correct offset.  When I issue G0 X0 Y0 F500 the engraver goes to my desired origin.

ecncshopInd, 2020-10-30:
If iam in inch mode, shud I prepend a G20 before the G10 command ? 
Will this work: 
(option1)
G20
G10 L20 P0

OR it has to be in same line : option2: G20 G10 L20 P0 ?

Does G10 ( which involves EEPROM write ) : resets the G20/G21 mode ?

Bcos : Iam facing some wierd issues when using G10 in inch mode. 

Iam using $13=1 to get back inch values from grbl. 
Will this $13 scale MPOS or WCO or both ? ( in response to "?" command ).