gnea/grbl #118: PWM output to inverter

poesel, 2017-02-04:
Caveat: this may not be related to grbl at all. Please close the issue if you think it's in the wrong place.

I've connected D11 to the 5v input on my Huanyang inverter. This works until you go below 50Hz. Then the spindle starts to jerk (no load). I guess this is caused by the PWM output of the Arduino and the inverter interpreting the 'long' 0v phases as stop. So I added a RC filter and now the spindle works down to 14Hz without jerking. 
Problem: when I use PWM control and move the gantry together then all axis start to rattle like crazy. Moving without spindle is ok, spindle without moving is ok but not botch. 

Im at a loss how the RC could influence grbl? How do you connect the Arduino to an inverter?

Thanks

jahnj0584, 2017-02-04:
Your inverter should be set to 400HZ at all times.

poesel, 2017-02-05:
Why would I do that? It is the point of using an inverter and grbls pwm output to have variable speeds. If I wanted only one speed I just need a switch.

chamnit, 2017-02-05:
@poesel: Not sure if this will help you or not, because I'm not quite sure what you are asking. You can alter the spindle PWM output frequency in cpu_map.h. It's currently at 1kHz for lasers. It can be altered to 2 other higher frequencies, if I remember right. Also, if you just want a switch, you can disable VARIABLE_SPINDLE which will move the pin to D12 from D11 (or you can alter this in cpu_map.h to stay the same).

closed 2017-02-05 by chamnit

poesel, 2017-02-05:
I wanted to control the spindle speed with the inverter and I found the solution in the meantime:
[image]
[image]
R2 4.7 kohm
R3 180 ohm
Rpot1 310 Ohm
C1 47 uF, 16 V
That may be useful for the wiki because the pwm output is not only useful for a laser.

Harvie, 2020-11-23:
Now there seems to be quite elegant solution integrating Huanyang RS485 communication directly to GRBL, i would be happy if you can help with testing:
https://github.com/terjeio/grblHAL/issues/68