gnea/grbl #121: Why doesn't it move smooth?

adibiut, 2017-02-07:
Hi. I have a laser cnc contlolled by Grbl v1.1f.
This is the video:
adibzaboli.ir/s.mov
Why doesn't it move smoothly? Is it normal?
Laser mode in on.

winder, 2017-02-07:
What gcode sender are you using?

adibiut, 2017-02-07:
@winder Grbl Panel

adibiut, 2017-02-07:
It seems that stops completely befor every change in rpm

winder, 2017-02-07:
The latest version of Grbl Panel should be fine with grbl 1.1. I'm not very familiar with laser mode so I can't really help there.

swarfer, 2017-02-07:
Are you sure $32=1 ?
You will have to post the Gcode so we can see it to make sure that is not the problem.
Laser mode tries to keep moving, but some Gcode will make it stop anyway.

adibiut, 2017-02-07:
[image]
@swarfer Yes I'm sure. this is the screenshot

adibiut, 2017-02-07:
@swarfer and this is the code.
[code.txt](https://github.com/gnea/grbl/files/757943/code.txt)

adibiut, 2017-02-07:
@winder Thanks.
It's ok with other codes which don't need to change the spindle speed. It runs smoothly.

adibiut, 2017-02-07:
@chamnit

chamnit, 2017-02-07:
@adibut :  You are commanding a spindle speed change without a motion while in M3 constant power mode. It has to sync and stop in this scenario. Spindle speed can be placed on the same line with G1 and you only need M3 once. 

This is explained in the wiki's laser page. Please read and understand this before posting next time.

closed 2017-02-07 by chamnit

mayhem2408, 2017-02-07:
What software are you using to generate the GCode.
I can see you issuing M03 commands to set the spindle sped, then issuing a move. You only need to set M03, or M04 once at the start of your laser burn and then issue moves and spindle changes together. You also don't need to issue the feed speed on every line if it is not changing. It will reduce the data transfers and depending on the grbl sender, it will utilise the serial buffer better. 

Your code looks like this
`G21
G90
G00 X0.0 Y0.0
M03  S63
G01 X0.5 Y0.0 F500
M03  S127
G01 X2.5 Y0.0 F500
M03  S159
G01 X3.0 Y0.0 F500
M03  S127
G01 X3.5 Y0.0 F500
M03  S95
G01 X4.0 Y0.0 F500
M03  S127
G01 X4.5 Y0.0 F500
M03  S95
G01 X5.0 Y0.0 F500
M03  S127
G01 X8.0 Y0.0 F500
M03  S159
G01 X8.5 Y0.0 F500
M03  S127
G01 X9.0 Y0.0 F500
M03  S159
G01 X9.5 Y0.0 F500
`

The optimised code would look like this
`G21
G90
G00 X0.0 Y0.0
M03 S0
G01 X0.5 Y0.0 F500 S63
G01 X2.5 Y0.0 S127
G01 X3.0 Y0.0 S159
G01 X3.5 Y0.0 S127
G01 X4.0 Y0.0 S95
G01 X4.5 Y0.0 S127
G01 X5.0 Y0.0 S95
G01 X8.0 Y0.0 S127
G01 X8.5 Y0.0 S159
G01 X9.0 Y0.0 S127
G01 X9.5 Y0.0 S159
`
If you really want to optimise the code, you do not need it issue the Y movement on every line if it is not changing.
`G21
G90
G00 X0.0 Y0.0
M03  S0
G01 X0.5 S63
G01 X2.5 S127
G01 X3.0 S159
G01 X3.5 S127
G01 X4.0 S95
G01 X4.5 S127
G01 X5.0 S95
G01 X8.0 S127
G01 X8.5 S159
G01 X9.0 S127
G01 X9.5 S159
`