gnea/grbl #123: Stock Controller Card Connecting Limit Switches

rseijo, 2017-02-10:
Hi,  I am new to CNC and GRBL.  Wondering if Limit Switches can be connected to the stock controller card?  if yes, does anyone have wiring and GRBL programming instructions?   Using GRBL v1.1f

The card in question is the Woodpecker CNC GBRL 0.9

Thanks!
Ralph

[image]

electrokean, 2017-02-10:
Start here: https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches
Although that is a work in progress - there is also a long discussion at #96
Nothing specific to the Woodpecker though

luben111, 2017-02-10:
@rseijo 
Can you bring some more information about your board? Some links, schematics and data sheets in order to help you. Thanks

It looks like Arduino core with added dedicated connectors for stepper drivers.

chamnit, 2017-02-10:
Closing because this is not a bug, but feel free to continue.

closed 2017-02-10 by chamnit

rseijo, 2017-02-10:
Thanks so much for the quick responses!!
In review of the Wiki for Wiring Limit Switches it covers Arduino Cards & Shields; not the stock board (in above image) that came with the CNC Machine.  Any ideas?

Also,  I have yet to locate any nomenclature, data sheets, etc. for the stock board.

luben111, 2017-02-11:
@rseijo Until this is Arduino nano processor with Grbl you can do the following:
- measure with multimeter and find which header pins to which pins of micro controller are connected
- once you have this list we can figure out the correct end switches wiring

electrokean, 2017-02-11:
Ralph, the only reference to this "stock board" you mention that I could find is one AliExpress seller:
https://www.aliexpress.com/store/product/GRBL-0-9J-USB-port-cnc-engraving-machine-control-board-3-axis-control-laser-engraving-machine/1941516_32713561151.html
This isn't really a common board, and if there is no documentation for it it will be difficult for us to assist you.

The standard grbl limit inputs are on D9 (X) D10 (Y) and D12 (Z).  See https://github.com/gnea/grbl/wiki/Connecting-Grbl
There are pins on the header that seem to be labelled A0-A7 and D13 - and the diagram on the above linked page tells you what those are used for.
I can also see some with labels starting with X/Y/Z but I can't make out the labels clearly.  At a guess I'd expect those to be the ones you need for the limits.  Can you tell us what they say?  Maybe a close up photo from front and back of PCB near that connector would help us.
If you have a multimeter, it would be possible to use that to trace out the connections from the ATmega IC to the header pins as suggested by @luben111 

All that said, this board uses a CH340 USB to serial chip, and this has been known to be the cause of communication problems for many people (data loss).  I would personally recommend getting a board that is more common and well documented, such as the Protoneer shield, and an Arduino Uno to plug it into.  I expect those stepper drivers on the Woodpecker board follow the standard "Pololu" pinout, so you should be able to move them across to a new shield.  After all the time and cost invested a CNC, you really don't want to use an unreliable and unsupported controller, even more so as a beginner.

BlueOrangeLive, 2017-02-11:
Hello.

The external contacts are all GND.
2* Limit-X-Axis
2* Limit-Y-Axis
2* Limit-Z-Axis
A5 = Probe
A4 = 
A3 = Coolant Enable
A2 = Cycle Start/Resume 
A1 = Feed Hold
A0 = Reset/Aboat
A7 = 
A6 = 
D13 = Spindel Direction
RST
5V

[Limit Switch Wiring:](https://github.com/gnea/grbl/issues/96)

From another shop's view from the back of the board:
[image]

Greetings Jürgen

luben111, 2017-02-11:
We need to know the connections between the header pins and the microcontroller. 

Use a multimeter and probably a needle or pin to touch the microcontroller pins. It may take 15-20 minutes to get all connections.

rseijo, 2017-02-11:
Kean, Luben111 and Jurgen, thanks so much for the guidance and collaboration!   I thought I was going to be lost on this issue.

Jurgen, I think you have cracked the code 👍  I will try the suggested setup and report my finding.

Incidentally, I order an Arduino UNO with the CNC Shield, so I have that as an alternative if this stock board does not workout,

BTW, here is a closeup image of the stock board Front.
[image]

Thanks again!

luben111, 2017-02-12:
@rseijo You still need to measure which pin of the header to which pin of the microcontroller is going by using simple multimeter. Set the multimeter to measure short circuit (beeps when you touch both probes). See below images describing the sequence. At the end you should get some list as:
Pin1 (Xon1)  - pin XX of Atmega328
Pin3 (Xon2) - pin XX of Atmega328
Pin5 (Yon1) - pin XX of Atmega328
......
If we have this list we could draw nice diagram how to connect the limit switches.
[image]
[image]
[image]

rseijo, 2017-02-12:
Hi luben111,  I will check that out and report back findings.  It will be later today; on my way out to the grocery story :-)   Thank you for the detailed instructions with images, this helps immensely!!!

rseijo, 2017-02-13:
I checked the PIN Out of the multi-meter and found that the Zen, Yen & Xen had no connection points to the Atmega328 chip.  Continued checking the other pins and have identified where they reside on the chip.

So with I enabled the $21 - Hard limits, boolean ($21=1) and connected a Normally Open switch.  Placed the GRBL (Candle v1.1.) in Jog mode to move the Y Axis, while moving the Y Axis closed the switch and sure enough the Axis stopped moving.   Pressed Reset, Alarm appears, pressed Unlock and I am able to continue the Axis travel. 

Is this the correct response and sequence to restart the travel?

[image]

luben111, 2017-02-13:
@rseijo The end switches are maybe going to some serial resistors - see image below.
[image]
You need to find where the end switches are going, you can visually follow the wires on the PCB.

Also the enumeration of the microcontroller pin numbers is as follows (please add the pin numbers in the table where to each header pin is corresponding some controller pin). 
[image]

Without populating the table above with some data we can't make the correct wiring,

luben111, 2017-02-13:
BTW, if your current GRBL version is before 0.9j and you'd like to use the latest versions of GRBL you probably have to modify the PCB - the Z limit switch in latest versions is attached to different controller pin (compared to version before 0.9j). As I see the listed version is 0.9j so maybe there is no need to modify the board.

rseijo, 2017-02-14:
Hi Luben111,  I posted the updated diagram above and here too.

The PIN Outs for the X, Y & Z connections are on the backside of the PCB.  Added 2 close up pictures.

[image]

[image]

[image]

rseijo, 2017-02-14:
With regard to GRBL version, I am using Candle v1.1.7

[image]

electrokean, 2017-02-14:
OK, so it looks like
- there are two pairs of header pins per axis wired in parallel - perfect if you want two NO switches
- R20/R21/R22 are 100 ohm resistors in series with each limit input (for protection)
- there are capacitors in parallel with the inputs to filter noise
- the traces from those resistors disappear under the ATmega chip

Limit pins on the ATmega should be
Axis = Arduino pin name = AVR pin name = pin number via series resistor
X = D9 = PB1 = pin 13 via R20
Y = D10 = PB2 = pin 14 via R21
Z = D12 = PB4 = pin 16 via R22

With your multimeter you should be able to measure the 100R resistor between those header pins and ATmega pins to confirm

luben111, 2017-02-14:
@rseijo 
Now you need to make the final step and measure the connections between other end of resistors and the microcontroller as @electrokean  suggested.

[image]

After that we'll create the proper wiring diagram for your board

luben111, 2017-02-14:
The final wiring will be (first we need to confirm the connections - see above)
[image]

rseijo, 2017-02-14:
Hi Kean & Luben111,  
Thanks for the continued guidance!   Below is updated diagram with the PIN Locations for R20/R21/R22 on the Atmega328 chip.

Luben111, the diagram with the switches is awesome.  Thanks 

[image]

luben111, 2017-02-14:
@rseijo 
Well done! Now we have the whole puzzle:
pin 13 in TQFP32 is PB1 - goes to pin15 of DIP 28 and 
pin 14 in TQFP32 in PB2 - goes to pin 16 of DIP 28
pin 16 in TQFP32 in PB4 - goes to pin 18 of DIP 28

This means that the wiring above is correct and you can use to connect the end switches. 
Also this shows that you can upload  GRBL v.1.1 and everything should work.

BTW, the Woodpecker board is not bad at all and I'm inclined to use it in the future for small projects.

electrokean, 2017-02-14:
@rseijo great work - if only they labelled it better, or bothered to provided documentation
@luben111 what a pity it uses a CH340

rseijo, 2017-02-14:
Luben111,  Will do!   Thanks for the confirmation and all your help!!

rseijo, 2017-02-15:
Guys, Complete and Total Success!   
Connected the 4 NO switches (X & Y) and everything stops (travel and spindle) when a switch is triggered.  Works perfectly!  I need secure the wiring and then move on the next challenge;  come up with a mounting bracket/place for the Y Axis switches that don't take up too much space.   

I'm so grateful to your support and guidance.  Best regards!

Ralph

rseijo, 2017-02-15:
Call me crazy, any suggested resources of where I can source parts to make a large CNC?  1ft x 3ft Table.

electrokean, 2017-02-15:
That's pretty open ended - there is many types of CNC, but I'm assuming you mean a gantry style.
Also 1ft x 3ft isn't that large - maybe for a hobby user, but it is quite small for a business.
You could start with a common Chinese machine - do a search for "6090 cnc" on ebay or AliExpress
You're unlikely to get all the parts individually for much less, and there are sellers who supply the hardware without motors or electronics if you want to customise it.

electrokean, 2017-02-15:
Also, check out cnczone.com if you haven't already - lots of build logs there.

chamnit, 2017-02-15:
A shapeoko 3 XL is pretty close to those dimensions.

langwadt, 2017-02-15:
@rseijo parts are just a small part of building a cnc, but I've gotten stuff from http://www.robotdigg.com they've got tons of stuff and I think very decent prices

swarfer, 2017-02-16:
@rseijo http://www.openbuilds.com  get an Ox or C-Beam kit

cavoke, 2017-03-08:
@luben111 I'm currently having troubles with this board as well. I believe it is due to the CH340 chip. GRBL Candle does not respond, and UGS states "GRBL not finished booting". As I can tell this is indicating communication issues to which re flashing GRBL has helped other people solve this issue for other boards. As you mentioned above you could be able to flash GRBL 1.1 to this board.

Do you have any information on how to do this or any suggestions to help me achieve re flashing this board with either 1.1 or 0.9? I do have the hex files I'm just not sure what to do with them.

luben111, 2017-03-08:
@cavoke If the USB reflashing is not working -  check if drivers were installed correctly and reboot the computer. If this is not helping the only one available option is to reprogram the ATMega328P on the board. If you have AVR programmer or you can borrow it you can simply reprogram the board, not sure if the board has ISP header.

VECKOFF, 2017-03-09:
Hi Rseijo,
I have the same CNC (router + Laser) with the Woodpecker CNC GRBL 0.9, but the program is different (old version  than Candle 1.1.7, GRBLCONTROL 0.8).
I download and instal Candle 1.1.7 but the "control manual" don´t work
Can you help me ?

Thanks

robhiggins, 2017-03-11:
@cavoke  I have the same issue with my 2418 I just received last week. The board came unflashed I believe as there is no response from the board what so ever. The RxD LED does response to send requests from the serial port, but no TxD activity at all. CH340 chip makes connection, but no response from AVR at all.

VECKOFF, 2017-03-12:
Hi Robhigins,
Do you try with the GRBLCONTROL 0.8, see: 
https://github.com/Denvi/Candle/blob/master/readme.md
With GRBLCONTROL 0.8 my 1610 work fine

samuelketchup, 2017-03-22:
@cavoke 

I have the same machine and I'm not having the issue where it will not finish booting.  Could you please upload the HEX files so I can try a reflash? If you still need help, you can get Xloader from here: http://xloader.russemotto.com/
and select the hex file, Uno(ATmega328), Com#, baud rate 115200

pswarman, 2017-03-31:
Guys, THANK-YOU so much for all your work on limit switches and XLoader, I also have this board and have struggled with the lack of documentation. I'll now implement your design suggestions on the switches. THANKS. Paul.

marcams, 2017-05-06:
I received my Linkpsrite CNC1610 last month and was able upgrade it with limit switches and flash the new firmware from GRBL0.9 to GRBL1.1f. Homing function is working fine now (I'm using bCNC). The new firmware was flashed using Arduino IDE and configuring the board as arduino nano. Thanks guys for the references published here! ;-)

piggei, 2017-05-23:
> I received my Linkpsrite CNC1610 last month and was able upgrade it with limit switches and flash the new firmware from GRBL0.9 to GRBL1.1f. Homing function is working fine now (I'm using bCNC). The [...]

Did you ignore the alert about few memory for variables? I compiled the 1.1f but I didn't send the code to my card because I read that warning...

komradebob, 2017-06-21:
marcams,

When you compiled grbl 1.1 for the woodpecker, what board type did you use? Any other changes from the stock grbl?

Thanks!

micha88, 2017-07-21:
I had the Problem that my GRBL Board had no Arduino Bootloader installed. So I was not able to update the Board via USB.
As this Board has no ISP connector I had to flash the bootloader in a direct way (connect the pins directly on the Board).
After that step I am able to flash the Firmware over USB via Arduino IDE.

uberiain, 2017-07-21:
Hello, I have the same controller , but, when I upgrade grbl 0.9j to grbl 1.1f lost the configuration of grbl. Can anyone post the $$ parameters, please?. 

I have one 1610 cnc machine .https://es.aliexpress.com/store/product/2015-New-GRBL-3-Axis-Pcb-Milling-cnc-Machine-1610-Diy-Wood-Carving-Mini-Engraving-Pvc/1941516_32553569537.html

Brandon2255p, 2017-07-27:
I have bought the 3018 but the parameters may be the same 

$0	10	(step pulse, usec)
$1	25	(step idle delay, msec)
$2	0	(step port invert mask:00000000)
$3	5	(dir port invert mask:00000101)
$4	0	(step enable invert, bool)
$5	0	(limit pins invert, bool)
$6	0	(probe pin invert, bool)
$10	3	(status report mask:00000011)
$11	0.010	(junction deviation, mm)
$12	0.002	(arc tolerance, mm)
$13	0	(report inches, bool)
$20	0	(soft limits, bool)
$21	0	(hard limits, bool)
$22	0	(homing cycle, bool)
$23	0	(homing dir invert mask:00000000)
$24	25.000	(homing feed, mm/min)
$25	500.000	(homing seek, mm/min)
$26	250	(homing debounce, msec)
$27	1.000	(homing pull-off, mm)
$100	800.000	(x, step/mm)
$101	800.000	(y, step/mm)
$102	800.000	(z, step/mm)
$110	800.000	(x max rate, mm/min)
$111	800.000	(y max rate, mm/min)
$112	500.000	(z max rate, mm/min)
$120	10.000	(x accel, mm/sec^2)
$121	10.000	(y accel, mm/sec^2)
$122	10.000	(z accel, mm/sec^2)
$130	200.000	(x max travel, mm)
$131	200.000	(y max travel, mm)
$132	200.000	(z max travel, mm)

I am getting a get_sync() error. Did you upload it using USB or an ISP connector?

uberiain, 2017-07-27:
Thanks , I uploaded grbl 1.1f with usb  without problems.

dneuman1, 2017-07-28:
@ micha88  can you detail what you had to do ... I am having the same issue 

"I had the Problem that my GRBL Board had no Arduino Bootloader installed. So I was not able to update the Board via USB.  As this Board has no ISP connector I had to flash the bootloader in a direct way (connect the pins directly on the Board). After that step I am able to flash the Firmware over USB via Arduino IDE."

Brandon2255p, 2017-07-28:
@dneuman1 I can help you. I literally just updated my board from 0.9 to 1.1f in the last hour. I would really recommend that you do not even attempt this if you do not have very accurate soldering skills, knowledge of arduino and either a spare arduino or an ICSP.

dneuman1, 2017-07-28:
@Brandon2255p Great!  I repair classic arcade PCBs as a hobby, so I have excellent soldering skills and spare boards

Brandon2255p, 2017-07-28:
@dneuman1 Just a couple pics that I took which show you my dodgy setup that actually did not short anything out:
[image]
[image]

komradebob, 2017-07-28:
I don't see anything dodgy about that...Well the clear tape maybe. :)

dneuman1, 2017-07-28:
@Brandon2255p - Cool! ... thanks for posting these pics of your setup

eye4tech, 2017-08-17:
Hi @dneuman1,

Thanks a lot for all the information and the pictures. They were very helpful and the soldering was easier than I suspected. I tried to re-flash the bootloader on my Woodpecker CNC board based on your information and using the methods provided by Arduino  [Arduino as an ISP](https://www.arduino.cc/en/Tutorial/ArduinoISP) and those by “Atmega bootloader program by Nick Gammon” ([Instructions](http://www.gammon.com.au/bootloader) & [Source on Github](https://github.com/nickgammon/arduino_sketches/tree/master/Atmega_Board_Programmer)).

Neither method worked for me. The Message I got was:
_Attempting to enter programming mode ......................................................
Failed to enter programming mode. Double-check wiring!_
I multiple times check the wiring and use genuine and Arduino clones to programme but without success. I was wondering how you can add the GRBL 1.1f directly without installing the bootloader as you did.
All insights are welcome as I need GRBL 1.1e or above for the software I would like to use on my 3018CNC laser cutter :-)
Kind regards
K.

PS: Just a last reflection,  I used two different USB power supplies ( computer "_A-ISP_" and independent wall socket "_Woodpecker CNC PCB_") so might not have a common ground (ref). Could that be the cause?

Brandon2255p, 2017-08-17:
Hi @eye4tech 
I would suggest that you use one power source. I realise now that I did not show how I powered it in the zoomed in pic. But if you look at the bigger picture you will see that I am using the arduino nano clone's 5V USB power to power the Atmega on the GRBL board. I also connected the common ground reference. I did this through the header pins because it was easier than attaching more to the tiny pins. But since you found soldering it quite easy then look up the pinout of the atmega328p to see where the power pins go. 

And I had trouble flashing the hex  file so I just cloned the repo and opened it using the Arduino IDE. You click Ctrl-shift-U in the IDE to upload using a programmer

eye4tech, 2017-08-17:
Hi Brandon2255p,
Thanks for that. Happy to see possibilities of solving this are not yet exhausted :-). Will try and let you know how it all goes :-).
Thanks for the quick reply.
K.

jasmattkrz, 2017-08-17:
Thanks to everyone for adding to this conversation, I just got this machine and am very interested in adding endstops.  I'm new to GRBL but I have a fair amount of Arduino and 3D printing experience and am willing to try soldering onto the 328p, but is the firmware upgrade necessary from 0.9j to 1.1 just for endstop functionality?  I've tested connecting the Xen, Yen, and Zen pins to ground and there is no effect with the current firmware (using Candle 0.9), so that leads me to believe it is.  But it feels odd that there are endstop pins built into the Woodpecker board, but they don't currently function.  

Any help appreciated!  This thread has been a great resource.

komradebob, 2017-08-18:
My woodpecker board (came with a LinkSprite CNC) is 0.9 and notices when I hit my newly installed limit switches.

jasmattkrz, 2017-08-18:
Thanks for the reply @komradebob !

And you're just using a two wire NO switch?  So a jumper between Xen (for example) and the ground pin below it should simulate that behavior? 

Are you using Candle?

komradebob, 2017-08-18:
Yup. That's exactly it. 

I'm driving it with Chillipeppr. 

Not sure exactly which version is on the board, its home, I'm not. :)

jasmattkrz, 2017-08-18:
Since you confirmed that the issue wasn't (physically) on my end, I dug back through these posts and I stupidly didn't have hard limits enabled!  I just had to type $21=1 into the console.  I got confused because Brandon2255p listed his as $21=0.

Forgive me, I am new to GRBL.  But my thanks everyone here and to @komradebob and @rseijo in particular!!

samsagaz, 2017-08-21:
I have the same board. Tested with 0.9 and 1.1 firmware and always get errors when use 5.5w laser. If use 2.5 or 0.5w laser everything works great. Someone know how is that possible?

eye4tech, 2017-08-21:
Hi Sam,

I have a 5.5 W laser 540 nm and can use it without problems. Mine has an
extra board that is attached between the laser and the board but not sure
about its function.
Have upgraded this weekend to GRBL 1.1f and all seem to be working fine.
What is the error you get? Do you have more detailed information?

Regards
K

[127 more replies]