gnea/grbl #125: Positive coordinate space after homing

classbproject, 2017-02-10:
Hello. I have a couple of questions around homing and some issues that I've been facing. I have two NC limit switches wired in series on each axis. I'm currently using only X and Y axis since this is a laser engraver. After some trial and error with $23, $5 and motor wiring flips, the machine homes to the top right corner [as shown [here](http://i.imgur.com/7hx7aED.jpg)]. I read in the Wiki that homing sets the machine in the negative coordinate space. This means that in the current location, a movement to the left would be -X and towards the bottom would be -Y. However, if I issue a command such as G0 X100 Y100, the machine moves correctly in the X-axis (which it should not since this is -X, but it does) but in the Y direction it tried to move further to the top and triggers one of the limit switches. 

How do I make the left and bottom of my workspace positive in both directions after homing? This would mean that left movement would be +X and bottom movement would be +Y. 

Also on startup, I see this:
```
Grbl 1.1f ['$' for help]
[MSG:Check Limits]
[MSG:'$H'|'$X' to unlock]
```
 even though #define HOMING_INIT_LOCK is disabled. 

Here are my changes to the config.h file:
`#define HOMING_INIT_LOCK` is commented out.
`#define HOMING_CYCLE_0 (1<<Z_AXIS)` is commented out.
`#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))` is enabled.
`#define HOMING_FORCE_SET_ORIGIN` is commented out (though I had enabled this earlier it didn't help with my issue. Hence I reverted it to its default state).

Current settings are: 
```
$0=10
$1=25
$2=0
$3=0
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=1
$24=100.000
$25=1000.000
$26=250
$27=5.000
$30=255
$31=0
$32=1
$100=80.000
$101=80.000
$102=80.000
$110=3000.000
$111=3000.000
$112=3000.000
$120=500.000
$121=500.000
$122=500.000
$130=300.000
$131=300.000
$132=20.000
```

luben111, 2017-02-10:
Hi,
GRBL works only with negative coordinates (X, Y and Z are negative digits) when homing enabled. You can't make it to work with positive coordinates until you have limit switches and homing.

chamnit, 2017-02-10:
@luben111 : Not quite. HOMING_FORCE_SET_ORIGIN will place the origin anywhere homing ends. So if you home to the front-left-bottom, all of the machine coordinates will be in positive space.

@classbproject : Please see this [wiki FAQ](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to).

closed 2017-02-10 by chamnit

classbproject, 2017-02-10:
I did read that but I still find it confusing. I did uncomment HOMING_FORCE_SET_ORIGIN in config.h and then homed the machine to front-left-bottom corner. Now the Y-axis seems to be in positive space but the X-axis is reversed. So when I enter G0 X-10 THEN the axis moves in the +X direction.

chamnit, 2017-02-10:
@classbproject : You need to invert the direction pin of your X-axis.

classbproject, 2017-02-10:
Ok did that but it homed the X-axis on the opposite side. I then inverted the X and Y directions for homing and now it seems to be in the positive coordinate space. Thank you for helping me sort this out.