gnea/grbl #126: Upgrade from 0.9 to 1.1, now not homing

RobDrizzle, 2017-02-12:
Hi,
I recently upgraded from 0.9 to 1.1 for the laser mode. I didn't want to lose a working set-up so I simply used a different Arduino (Uno R1). The 0.9 was running on a Arduino R2. I swapped the arduinos and moved my Z limit to 12 and wired the laser to 11 and entered all of the existing settings as well as enabled the laser. Laser works fine but now the machine refuses to home. I get a "Homing Error" in Laser Grbl and an "Error 9" in Candle. I used to get a quick stepper "step" when connecting to the 0.9 set-up and absolutely nothing with the 1.1 version. Swapping back to the 0.9 arduino and changing the pins yields a perfectly working machine...

Limits are wired through external pull ups and caps for filtering. Like this but with .47uf caps across the pins to ground.
https://lh4.googleusercontent.com/-FhXABwKcb-I/UvMhBLTESlI/AAAAAAAADCI/kfE-f7jkrwc/s800/cnc_limit_switch_c3.png

Loosing my mind on this....

Here is the dump.

Dumped in Candle
[CTRL+X] < Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
$$ < $0=30
$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=3
$11=0.050
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=1
$24=25.000
$25=300.000
$26=5
$27=1.000
$30=1000
$31=0
$32=1
$100=320.000
$101=320.000
$102=320.000
$110=400.000
$111=400.000
$112=300.000
$120=10.000
$121=10.000
$122=10.000
$130=312.000
$131=360.000
$132=46.000
ok
$H < ALARM:9
ok
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
S0 < error:9

Dumped in Arduino
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
$0=30
$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=3
$11=0.050
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=1
$24=25.000
$25=300.000
$26=5
$27=1.000
$30=1000
$31=0
$32=1
$100=320.000
$101=320.000
$102=320.000
$110=400.000
$111=400.000
$112=300.000
$120=10.000
$121=10.000
$122=10.000
$130=312.000
$131=360.000
$132=46.000
ok
ok

? Call
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|WCO:0.000,0.000,0.000>
ok
ok

chamnit, 2017-02-12:
@RobDrizzle : Two questions. Do you have a two axis machine? Grbl is setup for 3-axis and homes z first. Alarm code 9 is thrown when Grbl moves the homing axis 1.5 times max travel and doesn't trigger the switch. 

Second, does Candle support Grbl v1.1? I'm not familiar with that sender.

RobDrizzle, 2017-02-12:
@chamnit Yes, three axis. Candle does work with v1.1. I'm trying out senders now to replace Grbr Panel which the new version doesn't like my computer... I'm also running LaserGrbl which runs v1.1
https://github.com/arkypita/LaserGRBL

The machine doesn't even attempt to move the z stepper.

chamnit, 2017-02-12:
Can you provide your $I build info strings? Also can you move all of your axes via a G0 or G1 command?

RobDrizzle, 2017-02-12:
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
[VER:1.1f.20170131:]
[OPT:V,15,128]
ok

$I

ok
[MSG:Caution: Unlocked]
ok

G1 X25 F25

ok
ALARM:2
[MSG:Reset to continue]

This happens no matter what axis I move.

Also, the homing error  takes about 20 seconds to return, it's not immediately after entering the command.

RobDrizzle, 2017-02-12:
Well... Above is not totally true...

G1 X-25 F25

The software "moves" the axis without error, but there is no motion on the machine.

chamnit, 2017-02-12:
Do you need your step enable pin inverted?

RobDrizzle, 2017-02-12:
No... well it wasn't inverted on the previous build. Either way I changed it to 1, cycled power, and tried. Same behavior, after sending the homing command nothing happens and I get the error after about 20 seconds

chamnit, 2017-02-12:
The reason why Grbl was giving an alarm when moving with G1 is because you have soft limits enabled and are trying to move outside the soft limits. When you move negative, it works. That's normal. 

Given the fact that Grbl is showing you the correct position after a commanded move and homing takes a while before it alarms out means that Grbl is pulsing steps correctly because Grbl tracks position by directly counting steps when they happen. I think your problem seems to do with something related to your stepper drivers not seeing the step signals. 

I'm not exactly sure why v0.9 works and v1.1 doesn't. Not sure if your other old Arduino has something to do with it either. R1 is fairly old.

RobDrizzle, 2017-02-12:
Swapping to a rev3 worked... not sure what the difference can possibly be. Are you using the Iref pin for anything?

chamnit, 2017-02-12:
No. IRef is not used. I'm not sure what the problem would be. I do know that R1 has some internal wiring differences on pin d13. And maybe updating the bootloader would help. Although likely unrelated, it has fixed some weird issues by doing so.

closed 2017-02-12 by chamnit