gnea/grbl #127: Maintain spindle function during homing [[ using spindle PWM signal as charge pump, Gecko G540]]

czuba, 2017-02-14:
Hi all,

So I'm looking for a way to maintain spindle function during automated maneuvers such as homing.

I'm using Grbl to control some stepper motors driven by a Gecko G540. The G540 doesn't have a separate enable/disable pin, leaving the motors to whine constantly & build up heat. However, it does have a 'charge pump' that expects a high freq. signal (≥10.5kHz) emitted from the control computer when it's ready to run, allowing it out of a fault mode and delivering power to the motors.

Turns out Grbl's variable spindle PWM output is a sufficiently fast modulation to trick the charge pump into acting like an enable pin. Bookending movements with:
```
m3 s10500    (spindle on @10.5kHz)
g4 p0.2    (brief pre-dwell)
(.....)
g0 x50 f6    (do the g-code stuff)
(.....)
g4 p2    (post-dwell)
m5    (stop spindle)
```
works great for single movements, but when I attempt homing with this arrangement, Grbl automatically disables the spindle while homing, so the G540 driver never exits fault mode & the motors never run.

I tried simply commenting out commands in the c-code that looked like they might be the problem, like  `spindle_stop();` in motion_control.c, but that doesn't seem to make a difference.

I'd expect other stepper drivers also expect a charge pump signal, so perhaps this feature would also have broader use & enable new Grbl applications.  ...for my particular case, I'm just using Grbl to control a set of motors & gantries, so there are not actual spinning cutting tools of destruction to worry about, but minimizing EM noise while the motors idle is important. 

Thanks in advance for the help and code development. This is a really awesome package and I'm psyched about what we'll be able to do with it.

---
_( Also, I realize the spindle speed is in RPM, while the charge pump wants Hz, but the unit-blind conversion value works, so I didn't feel the need to back out the true/effective cutoff )_

electrokean, 2017-02-15:
Although this could be done in software, there are limitations and trade-offs in terms of what can be added to the current code and still have it fit in a standard Arduino.  Even as a G540 user myself, I don't think this has a lot of benefit to the wider grbl community, who will mostly be wanting to drive an actual spindle or laser.
I'd suggest that it might be better to do this in hardware.  It would just involve a relatively simple 555 timer circuit, configured for the appropriate frequency, and the reset pin controlled by the motor enable signal (maybe via an extra transistor to ensure it "fails safe").
This has the added benefit of meaning you can stick to the standard version of grbl, and easily upgrade should there be any new releases.

chamnit, 2017-02-15:
@czuba : Grbl on the 328 is maxed out, so new public features are not likely on this build. However, I'll keep this in mind for the ARM version, which should debut shortly.

closed 2017-02-15 by chamnit

czuba, 2017-02-15:
Thanks for the quick feedback. I'll look into the 555 timer solution, and keep an eye out for the Grbl ARM.

langwadt, 2017-02-16:
as a compile time option is should be easy enough, just make the spindle functions empty and turn the timer on and off in the st_wake_up and st_goto_idle