gnea/grbl #129: firmware explanation

Eric-Hsia, 2017-02-16:
hi, everyone
     who can help me to explain below code: in planner.c line 433
     junction_unit_vec[idx] = unit_vec[idx]-pl.previous_unit_vec[idx];

thanks

MechaSteve, 2017-02-16:
This is calculating the difference in velocity at the junction. By taking the new segment vector and subtracting the previous vector, we get the vector pointing in the direction of acceleration. This is the acceleration that will change the previous velocity into the new velocity. (see diagram) The loop iterates over each axis to perform the complete vector math. 

[image]

This value is then used on line 445 where it is normalized (scaled to unit magnitude) and 446 to calculate the unit vector of the acceleration (scaled to remain under axis limits).

chamnit, 2017-02-16:
@Eric-Hsia : See [this](https://onehossshay.wordpress.com/2011/09/24/improving_grbl_cornering_algorithm/).

closed 2017-02-16 by chamnit

Eric-Hsia, 2017-02-17:
@MechaSteve @chamnit Thanks very much