gnea/grbl #130: Homing direction invert, mask problem

kefeur, 2017-02-17:
$23=0 :
Home scan :ok , start home ok. moving the machine max position .Good.

My settings $23=5.(x & y axis invert direction)
Start home scan function.
 Scanning the machine z ,yand x,y zero positions.
Start home function : GRBL  the moving x and y axis the max position (opps).
Is not zero, but the maximum position . Not well stored in the zero position.
The same is true if, after switching off ( machine current position is  x=+10mm & y=+10mm)
I started the home command , the end positions moving  will the machine , the axis max positions.
Please help .
Sorry my bad English.
Ferenc.

chamnit, 2017-02-17:
@kefeur : Read [this](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to)

closed 2017-02-17 by chamnit

swarfer, 2017-02-19:
yeah, invert Z and Y, according to the table at https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration you want the value '3'