gnea/grbl #131: Homing Cycle??

karimnageeb, 2017-02-17:
Hello Everybody,

First i thank you so much for such a very simple and amazing script. 

i just built my first Laser engraver. following this indestructible

http://www.instructables.com/id/Arduino-Laser-Engraver-Wood-Design/?ALLSTEPS

i got the machine build and wired up, am an electrical engineer myself. 

first the original design has the grbl 0.8J A little out dated so i flashed the newest V: 1.1f all looks good and i can see it from the terminal and control it. 

its laser engraver so no Z axis here. i don`t have driver or limit switches or anything related to the Z Axis. 

when i start the homing cycle i get alarms and errors Like 

Error: 9 
Alarm:8 

i ordered 3 other steppers Motors/ Drivers and a new arduino and made a little Limit switch board and hooked it up completely. 

on the new setup i started the homing cycle and i can see that the arduino activate the z axis first till it hits the limit switch and then activate the X, Y axis after words to calibrate the homing . then all good gives me an ok and i can actually send commands i can see the step motors drivers flashing executing the command.

how i can do the homing cycle without the Z Axis ?? no limit switches hooked up or drivers. 

i want the grbl just start with the X, Y and set the home position and go for the controls normally 

here is my built info :
[VER:1.1f.20170131:]
[OPT:V,15,128]

Here are my settings : 
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=3
$24=50.000
$25=500.000
$26=244
$27=3.000
$30=1000
$31=0
$32=1
$100=133.333
$101=133.333
$102=133.333
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=500.000
$131=400.000
$132=200.000

the settings is a mix between the instructable original design and reading through the new grbl wiki 
mostly setting $30, $31,$32 only modified 

any help would be appreciated i don`t want to wire a Z Axis . 

thanks a lot and excuse me being a newbie. 

Karim

wlaw, 2017-02-18:
> Wood-Design/?ALLSTEPS

Hello karimnageeb,

I'm just thinking out loud, but I think you should be able to fake the card
somehow.
My first thought is to use the direction pin for Z to drive a relay. Then
use the relay to short the limit switch for Z to ground. In this way when
the machine hits the limit the backs away then the process again. It will
think it has homed Z.

Not sure if this will work or not, but its a thought.

Bill

langwadt, 2017-02-18:
change config.h, make HOMING_CYCLE_0 the X and Y axis and HOMING_CYCLE_1 empty

chamnit, 2017-02-18:
Read [this](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#how-do-i-configure-my-homing-cycle-to-just-do-2-axes-or-i-have-a-pen-plotter-and-dont-have-a-z-axis).

closed 2017-02-18 by chamnit

karimnageeb, 2017-02-18:
Good Morning guys. 

i thank you all for your help. 

i followed your advice yesterday and edit the config file and re compiled the grbl. 

great news the machine ignores the z axis now as it should be. 

but i want to clear on something as i understand how the homing cycle happens. 

it activate the motor to drive the machine to one side then when hits the switch it stops and back off your switch pull off limit i think. 

this happens good with the X-Axis but on my Y-Axis it just have two little motions and stops doesn`t go all the way till hits the limit switch like the X-Axis and save the location. 

is this a normal behavior ?? or i have some of my settings wrong ? 

thanks for your help you are making my day now by finally seeing this machine moves