gnea/grbl #132: Y-Axis Homing ?

karimnageeb, 2017-02-18:
Good Morning guys.

i thank you all for your help.

i followed your advice yesterday and edit the config file and re compiled the grbl.

great news the machine ignores the z axis now as it should be.

but i want to clear on something as i understand how the homing cycle happens.

it activate the motor to drive the machine to one side then when hits the switch it stops and back off your switch pull off limit i think.

this happens good with the X-Axis but on my Y-Axis it just have two little motions and stops doesn`t go all the way till hits the limit switch like the X-Axis and save the location.

is this a normal behavior ?? or i have some of my settings wrong ?

thanks for your help you are making my day now by finally seeing this machine moves

chamnit, 2017-02-18:
@karimnageeb : No it's not normal. Please do everyone a favor and don't double post an issue. You likely have electrical interference causing a false trigger on your y-axis. There are plenty of online resources and solutions if you do a little research.

closed 2017-02-18 by chamnit

karimnageeb, 2017-02-18:
@chamnit thank you so much for your response. i didn't mean to do double post.
its the original post being closed like this one and i dont know how to open it again . 

for the electrical issue ill look into that but my limit switches on the Y-Axis are functioning properly to the pins on the arduino they open when they are not pressed and close to GND when they are touched. 

by the way am opening an issue here because i couldnt find the info i need while reading. i did read for the last two weeks good amount. issues and problems are not covered so i dont know if that behavior is normal or abnormal that`s why am asking.

excuse me being a newbie but i came to the source seeking information.

ill trouble shoot more and get back to you.

chamnit, 2017-02-19:
@karimnageeb : There has been a recent surge of homing and limit switch related questions, including several people emailing me directly. All of which were not a Grbl issue or bug, but an implementation problem by the user. Nearly all due to the lack of trying to figure it out on their own or bother reading the Wiki.

Honestly, I'm a bit tired of answering this rash of recent question, because it takes a lot of time to answer them. Most people don't realize or particular care that its my personal free time. Keep in mind that this is an open-source project. It's normal to have to figure out things on your own.

I recognize that you have tried and are trying. Thanks for that.

electrokean, 2017-02-19:
@karimnageeb As chamnit has said, this is a common problem and there are plenty of discussion and suggestions if you read through some of the other issue posts regarding limit switches like https://github.com/gnea/grbl/issues/96 (still open presumably so it is easily spotted)
Also electrical noise is mentioned in the Wiki page you were pointed to earlier as causing unusual behaviour - such as: [here](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#grbl-is-acting-very-weird-dropping-connections-freezing-dropping-lines-whats-up) and [here](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#when-setting-homing-pull-off-to-something-greater-than-zero-why-is-it-crashing-into-my-limit-switch ) - so I'd suggest studying all of that and applying some of the recommendations.

karimnageeb, 2017-02-25:
@chamnit @electrokean 

i thank you all for the great support and help 

i have a great update for you 

thanks to our discussion which guided me to the right way to pinpoint the problem 

1uF on the Y Axis limit switch fixed all the hustle so i added another on on the X-Axis to be safe too 

then i fired up and its homing as it should be the machine is working perfectly other than just one little thing which am reporting as am not being lazy or wasting time or don`t want to do my work 

just i don`t know whats normal and whats not. 

the machine was in negative coordinates as i understood this was the normal CNC code but i was using the Laser GRBL software i found here on this wiki and it always kept sending positive coordinates so i did more work and read through the wiki and learned how to fix that and i did fix it 

Just one thing is bugging me   i print Horizontally inverted on the X-Axis 

meaning to print a correct image i have to process the image through a software to make a mirror effect on the X-Axis just to print correctly. 

anyway. i just wanted to share with you what i have accomplished and just thinking with you guys to be directed on the right place where to look or how to start 

@chamnit  you my friend i appreciate everything you did and still do. this an open source and i do understand very bit of the word but am just seeking your generosity 

thanks again and here is one of my finished products 
[image]

electrokean, 2017-02-25:
See "$3 – Direction port invert, mask" in the [Wiki](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration)