gnea/grbl #133: Y Axis Cuts Upside Down

rseijo, 2017-02-21:
Hi Guys,
I finished wiring in the Hard Limits Switches and they work as planned!

I decided to run a G Code design (without the bit) to see how the Linkspite CNC performed. I uncovered two issues:
**1st.** The Y Axis is Cutting Upside Down and it moves to Y+ position to begin a cut that should start at ZeroXY position. (X Axis travels in the proper direction.) 

When I use the Jog commands the X, Y & Z axis travel in proper direction.   Note,  I did have to set the **Direction port invert, mask** to $3=7  (see below image) to correct travel when using the Jog.

**2nd.** The finished product is tiny. Something that 4.6" wide (X) by 2' tall (Y) is compressed to half the size.

Is this a result of having the Defaults for $100, $101 and $102 – [X,Y,Z] steps/mm?
$100=250.000
$101=250.000
$102=250.000
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm

I referenced the configuration instructions however, I am having trouble locating the Stepper Motor specifications "JLB Stepper Motor, Model 17HS1352-P4130" to calculate the
steps_per_mm = (steps_per_revolution*microsteps)/mm_per_rev
Also how does one determine the mm per revolution?

The Woodpecker Card (ATmega328 chip) is flashed to GRBL v1.1f  and has the A4988 Stepper Motor Drivers

**Direction port invert, mask** to $3=7 
[image]

Thanks Ralph

rseijo, 2017-02-23:
Update; please review and comment if this is appropriate or not.
Model T8 Lead Screw = 2mm Pitch.  The stepper motor spec is 200 steps per revolution 
Made the changes as follows to **calculated steps/mm for an axis**  
$100=800.408
$101=800.408
$102=800.408

Used following calculation method
[image]

langwadt, 2017-02-23:
why mix in inches  with a metric lead screw?  
200steps/2mm = 100steps/mm, 100 steps * 8 = 800 microsteps/mm

chamnit, 2017-03-05:
Closing due to inactivity.

closed 2017-03-05 by chamnit