gnea/grbl #134: Limit Switch Trouble

Manypopes, 2017-02-23:
I have an almost functional milling machine running GRBL. The only hurdle to overcome now is that my optical limit switches appear to be triggering on their own. I immediately assumed it was a noise issue, but checking with oscilloscope the signals appear to be good.

[image]
[image]
[image]

The oscilloscope signals seem to be consistently above 3V or less than 1.5V which correspond to the HIGH and LOW thresholds on Arduino.

My z axis always homes perfectly every time, but my x only works sometimes and y seems to trigger instantly most of the time. In that case the bed (y movement) will not move toward the switch at all, but will instantly jump away from it twice. Is this normal behaviour?

Is there anything else that could be causing issues here? I've configured individual axis homing and the homing per-axis fails in the same way.

I am using [https://www.ia.omron.com/product/item/2223/](these) optical switches. I have the signal wire connected to a 4.7kOhm pullup resistor to 5V.

I didn't record any data when the motors were running as it didn't seem to make a change, however I could go back and get this.

JBSchueler, 2017-02-23:
These optical switches are open drain. You did not mention if you are using pull-up resistors or not.
I would recommend 2k2 for each optical switch to let a current of 2mA flow thru the resistor.
Furthermore it is somewhat related to ticker #96.
If pull-up resistors does not ( totally ) solve the problem I suggest to add small capacitors close to the uP pins. Value of 10pF to 33pF will do.  Using bigger capacitors will make te measurement more stable but insecure as well.

electrokean, 2017-02-23:
Almost certainly a noise issue.  A very common and misunderstood issue.  Read #96 as suggested above.

Your scope traces show quite a lot of noise, even if you don't see pulses that go outside the thresholds.  The noise pulses in the limit cables is caused by sudden current flow changes in the motor or spindle cables, and can be very short (nanoseconds) and not easy to spot on a scope.  Your Z axis cabling might be routed differently, or you're just lucky there.

Add some capacitors, and report back.  I'd suggest 100nF capacitors, which are commonly available.  Values in the picofarad range aren't likely to be all that effective except on really high frequencies, and values above 1uF will begin to slow rise & fall times too much so you end up delaying the limit switch response.

If you don't capacitors handy, here is a possible test... What happens to the limits if you try home the machine but with the motor drivers disabled? - i.e. everything powered up and grbl thinking it is moving the motors, but they don't.  You'll need to manually trigger the limit switches at an appropriate time, before grbl times out on each axis.  If you don't see homing failures in this case, then that would point to noise from motor currents.

Manypopes, 2017-02-23:
Thank you for such fast replies!

I managed to track down some 100nF capacitors like you said and it seems to have worked brilliantly. I hadn't considered that the resolution on my oscilloscope would be too low.

closed 2017-02-23 by Manypopes

electrokean, 2017-02-23:
Great to hear.
Out of interest, what oscilloscope is that?  Looks like a USB connected one.  No indication of horizontal/time scale led me to the conclusion it is not a high speed one.

Manypopes, 2017-02-24:
It's a Hantek 6022BE, so I think about as low end as you can get. It has a 1MHz sampling rate and the graphs above are in milliseconds.

electrokean, 2017-02-25:
OK, not quite as low end as you can get - many cheap ones have bandwidths and sample rates way below 1MHz, making some barely usable even for audio.  In fact this one seems pretty reasonable for the price, as long as you're aware of its limitations.
You may have been able to capture the limit switch noise using the triggering function and a higher sweep speed.
FYI - there are some good discussions on the unit on EEVblog, like this: www.eevblog.com/forum/testgear/hantek-6022be-20mhz-usb-dso/