gnea/grbl #135: .cnc file format

Sajjadkhalil, 2017-02-23:
Hi, as I know the .cnc file contains supported gcode files grbl, so simply, I have changed the extension to .nc
and opened it with gcode sender, named, bCNC, Candle and winder, all of them visuales the file correctly, but after few gcodes sent it stopped, in the same location so I suspect that the problem in the gcode streamer.
the file I have used is attached
[test gcode.zip](https://github.com/gnea/grbl/files/796961/test.gcode.zip)

the problem that I want to use existing files that I am using which is in the extension of .cnc

gerritv, 2017-02-23:
Actually the fact that 3 GUI's stop in the same place means it is likely your gcode. What error is displayed in the Status response from Grbl? You can also try enabling Check Mode and seeing what error Grbl sends back without the steppers moving etc.

BlueOrangeLive, 2017-02-23:
@ Sajjadkhalil I have the file in bCNC looked at. But you must have a very large CNC because in the X direction the part is over 1140mm and in the Y direction 2360mm.
I have the file since about 20 minutes on bCnc and grbl 1.1 ... just instaliert without CNC machine further run time still about 1 stune and 40 minutes.

Greeting Jürgen from Germany I go to altweiber carnival, Helau.

Sajjadkhalil, 2017-02-24:
thanks friends for help,
In check mode the gcode works flawlessly, but when the program sending the gcode to the arduino it stops after few lines, I think that the gcode sender did not receive a message from the arduino to go further.
about my CNC I am trying to use it in stander CNC with dimensions of 2500 mm by 14000 mm, because I am tired of using desktop PC as it uses the NC studio.
no error is displayed, and will attach a picture of the test run.
 the only message I have got recently is Alarm:64 in bCNC app when performing scan edges
[image]
[image]
[image]
[image]
[image]
[image]

I have searched the meaning of alarm:64 in bCNC but I could not find any thing yet.

gerritv, 2017-02-24:
The message you received (Grbl 1.1 etc) is because your Grbl reset itself. You have electrical noise issues which is why it worked in Checkmode but not with motors and spindle running.

chamnit, 2017-02-24:
You are exceeding the max 30kHz step rate Grbl supports. Your y settings produce 66kHz. Reduce your max rate. 

Also your step pulse time is unusually large. Reduce that as well.

Sajjadkhalil, 2017-02-24:
thanks friends,
I have reuploaded the grbl to arduino using the precompiled .hex file instead of arduino IDE and it is now working
thanks chamnit for this note.

closed 2017-02-24 by Sajjadkhalil