gnea/grbl #138: 2 Pin Laser Setup

iceblu3710, 2017-02-25:
I have a power supply that requires a PWM 0-5V power setting (D11) set via the S command. This works as expected.

The M3/4 commands are my problem. On 0.9j I had "Spindle DIR" hooked up to the power supplies "Fire +" pin. In grbl1.1 the M3/4 changes controller motions but also modifies the spindle dir pin.

If I want to use M4 for smoother cutting then I need to invert the Spindle DIR pin manually. 

What is the proper way to set this up so that the "DIR" pin is only +5V on G1/2/3 ?

iceblu3710, 2017-02-25:
I put my meter on D11 and run a few tests and M3 & M4 do not change the S value depending on motion. 

These all result in D11 staying at 5V ALL the time
```
S1000
M3
G0 X10 F100
G1 X0 F100
M4 
G0 X10 F100
G1 X0 F100
```

Only `M5` changes D11 to 0v

chamnit, 2017-02-25:
@iceblu3710 : The operation of the spindle direction has not changed. It's been the same since for several versions now. What you want sounds like you want D13 to work as an enable pin. There is a config.h option for that. 

You also need to have laser mode enabled. See the configuration wiki page on how to do that.

chamnit, 2017-02-25:
Your multimeter will not likely pick up the quick transitions with laser power adjusting. The refresh is often too slow. You will need an oscilloscope or reduce your acceleration settings temporarily to something slow enough that your meter can pick it up.

closed 2017-02-25 by chamnit

chamnit, 2017-02-25:
Also, please use the current version, v1.1f.

iceblu3710, 2017-02-25:
@chamnit 

Can you quickly explain the proper operation for this setup.

I will upgrade to v1.1f and set my "laser fire" pin to D13. This will mean that when I set `M4 S1000` that D13 will only toggle on during `G1/2/3` and will go off for `G0`?

iceblu3710, 2017-02-25:
Working with:

D13 => L
D11 => PWM_IN

USE_SPINDLE_DIR_AS_ENABLE_PIN, INVERT_SPINDLE_ENABLE_PIN, SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED