gnea/grbl #140: Alarm code 2 with positive z move on laser

71tr6, 2017-02-26:
I got a small laser to play with.  It ran fine straight out of the box with Grbl 9.x that it came with.

I upgraded it with Grbl 1.1f, made the appropriate changes to config.h and installed limit/home switches at both ends of the x & y axis. The limit switches work fine and homing works excellent.

The issue occurs if a positive z move is encountered.  Since Grbl sees the natural machine 0,0,0 as the upper right corner, any attempt to move back, right or up from that corner is seen as out of bounds.

I'm hoping there might be a workaround for this since I'm using the z axis code to trigger the laser on and off.

chamnit, 2017-02-26:
Your options:
- Use v1.1's laser mode instead of hacking the zaxis to control your laser. 
- Turn off soft limits.
- Offset your z with work coordinates.

closed 2017-02-26 by chamnit

71tr6, 2017-02-26:
> Use v1.1's laser mode instead of hacking the zaxis to control your laser.
I didn't hack my z axis.  My CAM program doesn't do laser, just mill.  So, I wrote a program that inserts M03 (or M04) at a -z move and M05 at at +z move.  
I do have laser mode enabled but I don't see how that turns it on and off.  Perhaps you could elaborate?

> Turn off soft limits.
Rather defeats the purpose.

> Offset your z with work coordinates.
I like this, I should have thought of it.  I haven't made a post processor for it yet. It should be easy to put this into the header.

Thanks for you time

chamnit, 2017-02-26:
I would consider having to write a seperate program to turn on and off your laser a hack. Since you don't seem to use z axis moves at all other than to track when to toggle the laser, why keep them? It's just as simple to remove them as it is to insert an M3 or M5. 

There is a whole wiki page dedicated to the new laser mode. I suggest you take a read.