gnea/grbl #141: configuring grbl v1.1f

daruco, 2017-02-26:
Hi everyone and thank you for this great app. I'm just a single user trying to understand cnc. I bought a cnc router and I want it mostly to cut plywood for furniture design. It comes with a Arduino Uno board, nema 17 stepper motors, etc. The guy who made it to me said me "use grbl controller". So I'm here with some troubles about use it. Firstly, my computer is a macbook pro mid 2012. My Arduino is connected to my mac by usb cable. So I installed grbl controller v 3.6.1 wich comes with grbl v 0.8c. Everything was ok: finally after many times I can adjust the axis movement in order to move exactly what I want in millimeters. But stop button never functions. So I looked for help and i find this site. then I downloaded grbl v 1.1f, thinking "this must be better" but everything was worst: the axis direction is inverted, grbl settins dialog doesn't show anything, etc. So I wish you help me please configuring well this stuff, because i'm worried. I want to start to run a g code now, but this is frustrating to me: to understand how to configure perfectly my board and grbl controller Thak you, sorry about my english and i hope this is the correct place to ask.

winder, 2017-02-27:
For starters, I believe that GRBL Controller doesn't work with v1.1. You can try Universal Gcode Sender or bCNC which should both work fine on your macbook.

The bigger problem is that when upgrading GRBL from 0.8c to v1.1 you may have inadvertently wiped out the GRBL configuration for your machine. You'll have to spend some time figuring out what the correct settings are for your machine, the settings documentation is here:
https://github.com/gnea/grbl/blob/master/doc/markdown/settings.md

One of the settings, in particular, is `$2`, which you can configure to invert your axis direction back to normal.

daruco, 2017-02-27:
Thanks for answer. I'm going to read consciously the link you gave me.

chamnit, 2017-02-27:
@daruco : Also try reading the wiki. There is a lot information there that can help you.

closed 2017-02-27 by chamnit

vMeph, 2017-02-27:
Not sure whats your shield but pay attention on grbl wiki on versions 0.9 grbl above pins 11 and 12 of arduino was swap