gnea/grbl #142: EEPROM sometimes shows garbage or overrides after a successful run

paulusjacobus, 2017-02-27:
This is a rare issue but has happened 3 times so far. I run Grbl in laser mode to engrave a picture. The file is large say 25Mb and is feed through streamer.py. After the job finishes I can sometimes not start a new job. After inspection through the $$ command I file the eeprom filled with garbage. Resetting sometimes helps to get the arduino back in a normal state. Sometimes the eeprom is really overwritten with garbage and I have to re-enter all the values. After that re-entry, everything works fine again until I hit the same issue again. Hopefully you can advice?

chamnit, 2017-02-27:
Can you provide some more information?

- Your $$ settings
- Your $I build info output
- Provide the gcode program that is problematic. 
- Does this happen around the same spot of a job? Or is it completely random? And how often does it do it?
- Do you have an Arduino clone with a CH380 USB serial chip?

This sounds like something I can reproduce. So please provide everything listed.

paulusjacobus, 2017-02-27:
I will provide the three parts tonight when I get home:
Your $$ settings
Your $I build info output
Provide the gcode program that is problematic. (might be difficult to post 25Mb engraving file but will try)

I can shed some light on:
Does this happen around the same spot of a job? Or is it completely random? And how often does it do it? - It always happens when the job finishes, so the product, in my case the engraving completes. I speculates it's the exit routine that causes the issue.

Do you have an Arduino clone with a CH380 USB serial chip? I have an Arduino clone with a FDTI chip called FT231XS.

109JB, 2017-02-28:
I had this problem a while ago that i think resulted from exceeding a 30 kHz pulse rate.  I was testing to see how fast I could push grbl and had the stepping parameters, steps/mm, accel, and max feed rate, set such that I was well above 30 kHz. If I remember right I was in the neighborhood of 60 kHz and got EEPROM corruption.  I have been able to successfully run at 40 kHz but above that I would not trust.  Not saying this is your problem, but is a thought.

paulusjacobus, 2017-02-28:
@109JB  that might be well the case. What would be the max. Feed rate, acceleration rate to hit the 30kHz processing ceiling?

paulusjacobus, 2017-02-28:
My $$ settings are:
$0=10 (Step pulse time, microseconds)
$1=250 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=2 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=3 (Status report options, mask)
$11=0.200 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=400.000 (Homing locate feed rate, mm/min)
$25=600.000 (Homing search seek rate, mm/min)
$26=5 (Homing switch debounce delay, milliseconds)
$27=10.000 (Homing switch pull-off distance, millimeters)
$30=2100 (Maximum spindle speed, RPM)
$31=0 (Minimum spindle speed, RPM)
$32=1 (Laser-mode enable, boolean)
$100=315.500 (X-axis travel resolution, step/mm)
$101=315.500 (Y-axis travel resolution, step/mm)
$102=0.000 (Z-axis travel resolution, step/mm)
$110=10000.000 (X-axis maximum rate, mm/min)
$111=10000.000 (Y-axis maximum rate, mm/min)
$112=0.000 (Z-axis maximum rate, mm/min)
$120=500.000 (X-axis acceleration, mm/sec^2)
$121=500.000 (Y-axis acceleration, mm/sec^2)
$122=0.000 (Z-axis acceleration, mm/sec^2)
$130=330.000 (X-axis maximum travel, millimeters)
$131=230.000 (Y-axis maximum travel, millimeters)
$132=-0.000 (Z-axis maximum travel, millimeters)

My build data:
> $I
[VER:1.1e.20161219:]
[OPT:VZ]

I cannot upload the 25Mb file that I was using because Github only allows files up to 10 Mb.
However I can show a similar file:
[arduino_0003_Gray_256_gcode.txt](https://github.com/gnea/grbl/files/805954/arduino_0003_Gray_256_gcode.txt)

chamnit, 2017-02-28:
Yeah. It's too high step frequency. You're running at 52 kHz. Way above the max rated 30kHz. Above this value Grbl can start to do weird things. When it does is dependent on a lot of variables. I'm fairly surprised that you were running this fast successfully most of the time.

electrokean, 2017-02-28:
Based on your steps/mm settings, you need to reduce your maximum X & Y rates from 10m/min to roughly 5.7m/min (30000 Hz x 60 sec/min / 315.5 steps/mm = 5705 mm/min)
You will probably get away with 6m/min, maybe 7.5m/min based on @109JB comment above.

What feed rates do you actually use in your programs?

paulusjacobus, 2017-02-28:
@electrokean and @chamnit I usually use for laser cutting really slow rates like 300mm/min or even lower. For engraving I use about 600mm/min or max 900mm/min. Above that i see motor stutter. So one fix is to go back from 32 to 16 micro steps on my step sticks (drv8255) and see how well it does do the engraving. My engravings are 380dpi or 15pixels per mm. Below that resolution, i loose too many details. Everyone thanks for the responses.

electrokean, 2017-02-28:
@paulusjacobus did you edit your $$ settings list above?  I'm sure it had $110 & $111 as 10000 mm/min.
Editing factual details (rather than typos) can make the ongoing discussion confusing, especially when they're the focus of discussion.

I don't think grbl does any calculations using the max rate settings, and it just uses it as a limit, so I'm not sure this explains your original issue.  In any case if you never run above 900mm/min then I'd suggest you decrease those to something like 1000 mm/min to be safe.

paulusjacobus, 2017-02-28:
@electrokean yes I did. I was playing with the settings in Grbl yesterday but did first encounter the issue at 5000.00mm/min setting. I changed it in the Arduino to try to replicate the issue faster in case it was feed rate. Seems like I was already on the right track. However the eeprom did not get overwritten at 10,000mm/min with one attempt. So may be it is the feed rate in the g code that corrupt the eeprom at run time. I use cnc cheton as my gcode sender. It might be this node js tool sending the corrupt information. However it is very difficult to replicate so will keep an eye on it and try to see if I can run the engraving on a lower step rate. If the eeprom corrupt happens again then I will reopen this issue.

closed 2017-02-28 by paulusjacobus

swarfer, 2017-02-28:
@paulusjacobus  you can calculate the step rate at http://swarfer.co.za/cnc/grblcalc.html

vMeph, 2017-02-28:
To know at what Khz is running what would be the equation parameters to use?

109JB, 2017-02-28:
kHz = (steps/mm) * (mm/min) / 60 /1000

For example, my machine has about 315 steps/mm and I run at 5000 mm/min max feed rate

315 * 5000 / 60 / 100 = 26.25 kHz

When I was doing my testing I believe I just increased the microstepping which resulted in about 630 steps/mm and would have given:

630 * 5000 /60 / 1000 = 52.5 kHz

109JB, 2017-02-28:
> @paulusjacobus you can calculate the step rate at http://swarfer.co.za/cnc/grblcalc.html

That appears to only work for a machine with belt driven XY and screw drive Z

paulusjacobus, 2017-02-28:
@109JB and @swarfer thanks for the tip. I see that my settings should work (26.25kHz) but it is at the edge of the processor's capability, so will keep testing. Btw My machine is a k40 laser which is belt driven.

vMeph, 2017-02-28:
@109JB   my machine is set with 32 microsteps and my (mm/min) is at 2000
160×2000/60/1000=5.3khz
So i guess best choise would be (mm/min)set at 10 000 for y and x ?
160×10 000/60/1000=26,6khz
And z what would be best?
How about ( acceleration, mm/sec^2) settings i have default ones that comes set with grbl,does that needs any adjustments?

109JB, 2017-02-28:
The accel won't affect the max kHz. Only the step/mm and mm/min affect the max kHz.  If your settings are different for each axis then you use worst case numbers

electrokean, 2017-03-01:
@vMeph Your max rate in mm/min and acceleration settings will be dependant on what your motors can cope with.  There is no point setting your max rate to 10000 if your machine can't do that.  Better to set it to something sensible, and if you run a g-code program which tries to use a higher feed rate it will be limited to what you set and (hopefully) the machine won't stall.
You just want to avoid having too many microsteps, or too high a max rate, if the total will give you a total step rate that exceeds 30kHz or so.
Settings for Z axis will generally be much more conservative than you X & Y, especially if you have a CNC with a heavy spindle.

vMeph, 2017-03-01:
@electrokean  i have  Nema 23 stepper motor 270 oz.in=1.9NM ,3A 23HS8430 bipolar with DM542 drivers, i have test at 10 000 max rate, then set the accel to 500, made a test machine seems to respond ok, the speed was so fast, off course i will slower down the accel cause is way to fast how it moves.

electrokean, 2017-03-01:
@vMeph nice!  I have similar motors, but I can probably only safely do about 2000 mm/min on my machine

vMeph, 2017-03-01:
@electrokean  just one question about the $0=10 (Step pulse time, microseconds), this have to do with drivers responce?
For dm 542 what would be the apropriate settings?

electrokean, 2017-03-01:
@vMeph Your pulse length needs to be long enough to match the input circuitry of the driver, and some drivers have slow-ish opto-isolators which require close to 10us pulses, maybe even slightly more.  At 30kHz, it can safely be anything up to 16us, and 10us works for the majority of applications using grbl.

I don't have a DM542, but a quick google search says it can handle pulses up to 200kHz and a pulse length as short as 2.5us.  But no reason to make it that short if running way below 200kHz.
http://www.omc-stepperonline.com/leadshine-dm542-digital-stepper-driver-2050-vdc-with-1042a-p-77.html