gnea/grbl #143: I want to connect Arduino Mega   [4 pins / stepper / per axis]  (look image) how ?!!!

savosik, 2017-02-27:
[image]

109JB, 2017-02-27:
Grbl only supports drivers that have step and direction inputs.

closed 2017-02-27 by chamnit

anonimou0, 2019-10-28:
I know this is a closed issue, but 2 and a half years have passed and I want to know if this is still valid.

Can I only use drivers with step/direction? Or can I use 4 pins, as previously asked?

Norwegian-Gadgetman, 2019-10-28:
GRBL doesn't have a way to keep the 'pin state' needed to run directly.
In fact, GRBL doesn't even know how many microsteps that each step is divided into.

If you want direct control, you will either have to rewrite GRBL, search for another package that does what you want, or write your own.