gnea/grbl #145: Z Limit switch doesn't work

Steve-Wellens, 2017-02-28:
The X and Y switches work fine.   I did some research and found this suggestion:

"I have solved my problem, I have to change Z limit pin to Spindle Enable for CNC Shield. " Z Limit pin and spindle enabled swapped to access hardware PWM on Pin 11."

I'm new at this and do not understand the solution.   Could someone explain it to me like I'm a slow witted 8 year old?  Thanx!

electrokean, 2017-02-28:
Arduino pins D11 and D12 were swapped in grbl 0.9 to gain access to additional functionality, so you may have to rewire those depending on the type/age of your shield.

See the top of the page at https://github.com/gnea/grbl/wiki/Connecting-Grbl where it is highlighted as a backward compatibility issue.

You could also recompile with the variable spindle feature disabled if your don't need that.  Better to just swap the pins by bending them out and running some wires as cross-overs.

Steve-Wellens, 2017-03-01:
Thanks so much!   It's working.

closed 2017-03-05 by chamnit

AcrimoniousMirth, 2018-11-10:
Hi,
Sorry to open up this issue again but I’m looking at a similar problem in 0.9.15 and wondered if this was the solution. Is there a way to swap the pin designations digitally and relatively easily? I’m using the GRBL Shield V3.0 and my endstops are wired NO.