gnea/grbl #146: It Doesn't Store parameter in E2PROM

arubioAtgithub, 2017-03-01:
I`m modifying some of the parameters but whenever I cycle the power or connect/disconnect the serial link the setting go back to the default ones.
I have tried $RST=* but without any success.

chamnit, 2017-03-01:
Haven't heard of this type of problem before. Can you describe your setup? Does Grbl report a read error prior to restoring defaults? Are you sure you are changing the settings? Meaning when you alter them, the $$ printout from Grbl shows the setting has been changed.

arubioAtgithub, 2017-03-01:
It seems to be related with the Arduino board that I was using. I switched to anoder one and Now it is working properly. Now When ever I try ot reprogram the old one I get some verification error.
Thanks for the soon response.

chamnit, 2017-03-01:
Thanks. Try wiping the EEPROM with the EEPROM clear example sketch in the Arduino IDE. You may have to alter the size of the EEPROM in the sketch. There a short writeup on this in the Wiki somewhere. Can't remember exact where at the moment.

closed 2017-03-01 by chamnit

arubioAtgithub, 2017-03-02:
I tried with the EEPROM Clear example from the IDE. I was unable to load it in the defective UNO board. In this way I found that the problem was related to the board and wasn`t a GBRL issue.
Thanks again

chamnit, 2017-03-02:
@arubioAtgithub : Thanks for following up and letting us know what the problem was.