gnea/grbl #147: Limit Switch Configuration

MechaSteve, 2017-03-02:
This is some general information for choosing a limit switch configuration and the advantages/disadvantages of each. We have had some good discussion on how to make GRBL work with various configurations, but little discussion as to why to choose one switch configuration vs another.

First lets discuss triggering and mounting of individual switches. Below are the four main ways limit switches can be set up.
[image]

The first configuration is the most simple. Normally Open switches (open circuit when not triggered) sense flags (object or feature that triggers switch) at each end of the travel of the axis. These switches may be easier to find and a triggering flag only needs to be mounted at the ends of the axis, where it will probably not be in the way.

However, this has the following disadvantages / failure modes.
If the power to the switches is removed or severed, the over travel will not be detected.
If any of the wires going to the switches are severed the OT will not be detected
If the switch is broken, destroyed or falls off, the over travel will not be detected.
If the flags are damaged, or out of position the over travel will not be detected.

The second configuration solves the first two problems. Because the switches are normally closed, if the power is removed, or the wires are cut; the switches will show an open circuit as though there is an over travel condition. It is important to note  that this does not make the switch work in these cases, but rather causes the machine to fault, forcing the operator to fix the issue. This is why NC switches are more commonly used for over travels.

The third case solves the third issue to some degree, and fully addresses the fourth. This configuration uses a flag that is present over the entire normal travel of the axis. This may be more mechanically difficult to arrange. This configuration does not address the first and second issues that the first configuration has, it also uses the lack of a signal to indicate normal operation. This is also why it only partially addresses the third problem. It will create an OT signal if the switch falls off or is out of position. But may not if the switch is damaged or destroyed.

The fourth case is the most robust and solves all four of the problems associated with the first configuration.

patgadget, 2017-04-24:
i would add that i choose NC because i had false triggering on limit while the motor were running.
In the NO mode, the signal is held HIGH with an internal resistor and could easily be interfere by electrical noise.
In the NC mode, the signal is held to ground (Solid) thus preventing any electrical interference to the input.
Pat

biasedlogic, 2017-09-07:
@MechaSteve - I would like to throw another aspect into the discussion about machine safety.
I've posted it here: https://github.com/gnea/grbl/issues/96#issuecomment-327947132 already. The thing is, we have still another choice to make and that's whether the switches are wired against ground (with pull-ups) or against power (with pull-downs).

langwadt, 2017-09-07:
I'd say that if you wire power to switches (or proximity sensors) at least make it a separate current limited supply.   Another thing to consider is that many micro switches specs a minimum current of several mA for reliable contact, the internal pullups in an MCU isn't going to deliver that

biasedlogic, 2017-09-08:
@langwadt exactly as you say. That's why I recommended in the linked comment not only limiting the current (a resistor will suffice for our applications), but also monitoring it (or the voltage drop across the resistor) to detect ground faults early. The thing about minimum reliable current is important too, it helps also avoiding interference from generating false inputs (low impedance circuit, i.e. one with a few mA flowing, is harder to distort than one with only few µA capacity)

MechaSteve, 2017-09-08:
So, my background:
Bachelors in Mechanical Engineering from Georgia Tech, and about 5 years working as an industrial systems integrator. I use a lot of industrial VFDs and Servo drives in my work. I also work with a lot of safety design for control systems.

**Limit switches are not safety devices:**
This is an important point. Limit switches are to, at most protect the machine. In most GRBL implementations this is especially important due to the risk of missed steps, belt jump, etc.

**Limit switches as homing devices:**
More ideally you would have both positive/negative over-travel switches, and a separate homing switch. The limit switches should be highly reliable, while the homing switch should be very precise.

**Proper Safety Circuit:**
In my CNC system I have a proper monitored safety relay. This directly interrupts the +24V supply to each stepper driver. GRBL is not safety rated, and should not be relied on for safety purposes. In my eventual design, the GRBL controller will only monitor the safety circuit. Any time the safety circuit trips, the controller knows it can no longer rely on the position of the steppers, and needs to re-home.

I need to properly document my CNC wiring diagram and post it here.

That being said, most desktop CNC and 3D printer designs do not need this level of robustness. In addition to my large 1000mm x 600mm x 200mm ball screw CNC, I also have a super cheap 200x150x50 CNC engraver and a 3D printer. I don't have nearly the same level of controls on the small machines.

biasedlogic, 2017-09-09:
@MechaSteve this sounds like really reasonable setup!

When you say "limit is not a safety device" and "grbl is not safety rated" you are very right, at the same time you touch something I was thinking a lot some time ago, when restoring an old coffee machine: Back then it was about grounding and protection against electrical shock. By modern standards you'd have to scrap that coffee machine altogether. It was built in a period when the norms weren't as strict and as well defined (it was built up to then-code) so I was between: 'it's not safe enough even to be powered on' and 'it was considered safe when it was new, if wired this way it's better than it was new'. So i put my effort to reduce risks, knowing well I won't get the modern standard of operational safety.

I think this is what we should preach in the hobby world: get most safety out of every penny. Know where it's not up to code and what's the remaining risk.

X3msnake, 2017-09-20:
@all 
thanks for sharing and helping others like me learn a bit more :)

Kim9999, 2018-04-11:
I built an opio isolator board to prevent noise works great

gerleimarci, 2018-06-08:
Hello,

I connected two NO switches per axis with 500Ohm pull-up resistor but after I send $21=1 to enable them, It responds with Alarm:1. I measured the 9,10 and 12 pin on the Arduino and all on 5V except when I push a limit switch. I tried to home and the Z axis did it well, but after that only the Y axis moved, the X didn't do anything. I tried to set $23 to 1. It started with Z, and now the X moved but just until the Y hit the limit switch and It gave the same Alarm:1. I thought there's a short between the X and Y limit switch. I measure the X(pin 9) and pressed the Y limit switch but it was all time high. 
What did I wrong?

17:57:47.539 $0=10
17:57:47.539 $1=25
17:57:47.539 $2=0
17:57:47.539 $3=0
17:57:47.539 $4=0
17:57:47.539 $5=0
17:57:47.539 $6=0
17:57:47.539 $10=2
17:57:47.539 $11=0.010
17:57:47.539 $12=0.002
17:57:47.546 $13=0
17:57:47.546 $20=1
17:57:47.546 $21=1
17:57:47.546 $22=1
17:57:47.546 $23=1
17:57:47.546 $24=25.000
17:57:47.546 $25=500.000
17:57:47.550 $26=250
17:57:47.550 $27=5.000
17:57:47.550 $30=1000
17:57:47.550 $31=0
17:57:47.550 $32=0
17:57:47.550 $100=160.000
17:57:47.550 $101=160.000
17:57:47.587 $102=480.000
17:57:47.587 $110=1000.000
17:57:47.587 $111=1000.000
17:57:47.587 $112=1000.000
17:57:47.587 $120=20.000
17:57:47.587 $121=20.000
17:57:47.587 $122=20.000
17:57:47.587 $130=500.000
17:57:47.587 $131=370.000
17:57:47.590 $132=200.000

Kim9999, 2018-06-12:
Check all of your wiring.   I installed switches but I also built the optio isolator board also.  I have pictures on my facebook. ( Kim Kottas in Montana) I wired my switches in series which is easier.   How did you wire yours?   It homes great.   I do have an issue with the soft limits so they are off at this time ($20).  
[image]  
[image] 
[image]

gerleimarci, 2018-06-12:
I wired the NO switches parallel. But my problem is not the noise because if I disconnect the limit switches' wires from the board(It has pull-up resistors), the problem does not disappear. So I think I have a software bug, but I don't know exactly what's the problem.

Kim9999, 2018-06-24:
Reflash your firmware and start over..But first disconnect it all and see
if problem remains,,,if not it is in the wiring of your system.