gnea/grbl #149: Z-limit switch not recognized

SteBuTOS, 2017-03-05:
Hi all,
the z-limit switch is recognized by the status report (see down below) but nut during homing: Z-axis moves downwards to surface, triggers the z-switch, but continous and crashes into mechanical limitations.
I would assume, that if the switches are wrong wired or have issues by interferences/ noises, they wouldn't become recognized by the status report.
Any help/hint is most welcome!
.........................................................................
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|WCO:1.000,-1.000,7.000>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:Z|Ov:100,100,100>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:X>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:Y>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:XY>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:XZ>
ok
ok
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=19
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=160.000
$101=160.000
$102=2560.000
$110=800.000
$111=800.000
$112=15.000
$120=10.000
$121=10.000
$122=10.000
$130=850.000
$131=850.000
$132=150.000
ok

chamnit, 2017-03-05:
By default, the z-axis moves up in the positive direction to search for the limit switch. I think you have your coordinate system backwards.

SteBuTOS, 2017-03-05:
Ok, understood. But even if its the case, shouldn't it stop anyway when it hits the limit switch?

SteBuTOS, 2017-03-05:
...from my understanding z-min and z-max share the same IO-Pin, so from grbls-view they are not different. Grbl initates z-movement in some direction until some z-limit switch is triggered.

chamnit, 2017-03-05:
It's always good to have a properly configured coordinate system. Things can behave not how you expect. Also provide your $I build info string, describe any changes you made, and how you are uploading Grbl.

SteBuTOS, 2017-03-05:
Here is my $I and... to exclude compile issues as possible root cause (which I first did) I know only use the precompiled hex files as downloaded from here.
.................
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|WCO:1.000,-1.000,7.000>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:Z|Ov:100,100,100>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:X>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:Y>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:XY>
ok
ok
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|Pn:XZ>
ok
ok
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=19
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=160.000
$101=160.000
$102=2560.000
$110=800.000
$111=800.000
$112=15.000
$120=10.000
$121=10.000
$122=10.000
$130=850.000
$131=850.000
$132=150.000
ok

SteBuTOS, 2017-03-05:
Some more thought upon proper configured coordinate system:
I use UGS as sender, when I press the "scroll down" button the z-axis moves down as I would expected it to behave. How can this seem to be correnct when it seems to be incorrect from the homing perspective?

chamnit, 2017-03-05:
Again, by default, Grbl searches for the homing axes in the positive direction. You have to explicitly alter the homing direction mask, if you want it to search in the negative direction. This is described in the wiki. In your case and provided settings, you haven't specified this.

If its moving in the negative direction already, that means Grbl detected your limit as already triggered and is trying to pull off the limit switch. This problem is fairly common and almost always caused by electrical noise.

SteBuTOS, 2017-03-05:
you where right. soldered in some 0.1µF capacitors between signal pin and ground and it worked.
Thanks a thousand times!

closed 2017-03-05 by chamnit