gnea/grbl #150: gitter room for GRBL  community

mkeyno, 2017-03-06:
hi , sorry to open ticket for my question , but as GRBL  community is truly expanded day by day , would it be better to open gitter chat room for normal question and bug?

chamnit, 2017-03-06:
@mkeyno : Thanks for the suggestion. I'm a total n00b with new web tools, as they also seem to appear day by day. I joined gitter, got a group and room going, but it's not immediately clear to me how this is going to help users get help.

neilferreri, 2017-03-07:
Maybe an official Google+ community?  Seems like a lot of the "issues" are really just the common problems anyone would have with a hobby level / DIY CNC machine.

gerritv, 2017-03-07:
But who will manage it? Sonny already has 2 github repositories to look at, with Arm version it will make in essence 3. He is already spending a lot of time managing these 'issues' many of which are really the result of not reading the provided information or doing some searching.

Gitter reminds me of XChat, no lasting information left behind. And many will avoid Google anything as it is just there to monetize you and your information.

jahnj0584, 2017-03-07:
Best bet is to link the solved issues in the wiki/configuring grbl for
future reference. If someone asks a repeat question you just give them the
section to read.

chamnit, 2017-03-07:
In that case, I need a much better homing operation document. About half of the issues are related to that.

closed 2017-03-07 by chamnit

neilferreri, 2017-03-07:
@gerritv That's my point with the community. Sonny and others doing the important work don't need to manage it or even be present. The community works to help and share with each other.  I'm sure it gets frustrating having to answer the 4000th question about electrical noise or negative machine coordinates all the while trying to have discussions about upgrading MCUs.  I don't know who would "manage" it, but I think the thing would run itself.

mkeyno, 2017-03-07:
its not really need to has ongoing online supervisor,  just like the other room every body cast its question and if any interested just share its solution, I saw many simple question answered quickly and ticket for real bug will be open in issue section, I see GRBL community is quite bigger than many other smaller lib follower community which is has its own room , @chamnit its really help you out to let the other answer ordinary question instead of open a ticket,  G+ and other social network are far beyond of gitter capability and proportionality , gitter its really fun and helpful and I saw many people immigrate to it from G+ and Facebook