gnea/grbl #152: Spindle motor won't work, but only from Windows, not Linux

Manypopes, 2017-03-09:
Hi,

I have a bizarre issue where I can control my spindle motor absolutely fine when sending commands from Linux but not from Windows, either through the Arduino serial monitor or Universal GCode Sender.

On Windows I can control the spindle at low rpm, but increasing it above about 4000rpm the motor will cut out. Once it has cut out, if I disconnect the usb cable it will work again up to maximum rpm, but plugging the usb back in will cause it to cut out again. If I leave it for a few seconds in its cut out state the warning light will appear on my stepper driver.

Any ideas why my m3 commands are being sent differently from Windows vs Linux? I've been using Linux while gettting the machine working but I intend to have a Windows computer for general users.

Would compiling and flashing GRBL from Windows make any difference? The version on it right now was uploaded from Linux.

chamnit, 2017-03-09:
@Manypopes : Grbl is OS-independent. So it should act the same, regardless of what is connected to it. The only difference would be the GUI implementation and electronic characteristics of the computer. My guess is you have some weird grounding/electrical issue with the Windows machine.

Manypopes, 2017-03-09:
It is a bit of a budget laptop, I'll see if I can test it out on something else.

electrokean, 2017-03-09:
This is likely to be a noise issue.  Spindle and stepper motors and their cabling get very noisy (electrically not acoustically) as they increase speed.  I'm assuming the same laptop is running both OS, so the Windows drivers for your USB to serial adapter could be less resilient to errors cause by EMI.
You need to implement some better grounding, shielding, and maybe filtering (ferrites on cables, capacitors across brushed DC motors and Arduino inputs).

Manypopes, 2017-03-10:
Hi, actually each OS is on a different laptop, so possibly the Windows laptop overall is less resilient. Might be interesting to boot Linux on it and see what happens then.

The spindle I'm using is brushless with a hall effect sensor, so plenty of wires coming off it, but they're all packaged in two shielded wires, one for the three phases and the other for signal wires.

I think firstly I'll try and find a different computer to use, if that doesn't help I'll try to reduce the noise.

electrokean, 2017-03-10:
If you have one, try a USB cable with a ferrite choke on it.
Google "usb ferrite" for images if you are not sure what I'm talking about.

closed 2017-03-11 by chamnit