gnea/grbl #156: Homing setup wiki page

chamnit, 2017-03-13:
To help alleviate some common questions, I've been trying to find some time to write a homing setup wiki page for weeks now, but I've been slammed with work on multiple fronts. If anyone is willing, I'd to get some help on get the homing setup wiki page going. It doesn't have to be pretty, but I think we can iterate to keep improving it. I've started a page and added some general points I'd like to cover. Thanks everyone!

jahnj0584, 2017-03-13:
Does the xcarve wiki have a page we can copy and edit ?

zentoolworks, 2017-03-15:
We would like to give it a try. 
Xin

Incredible-IT, 2017-04-06:
Homing is not the problem. My machine homes beautifully but what good is it if I then have to jog the machine to a arbitrary point and then set working position to zero. I can't find a definitive answer about what to do after homing. How do I set the origin? Some processes set a temp position that gets erased when the board is reset. EVERYBODY vaguely mentions some commands but they have been superseded or not explained. There seem to be commands like G28.1 that set something but what? Everything is in forums where one person will say something and 15 people will say something different, none of which work for me. My head is spinning and it's not just the beer...

jahnj0584, 2017-04-07:
You home the machine. You go to the origin for your work project, as chosen
in CAM. You then reset your work zero to set the origin. You then send this
code.

You want to disable G28 at start of program because it causes more trouble
then anything

chamnit, 2017-04-07:
The homing wiki page is a work in progress. Use the G10 command to alter the work coordinate system and to have it saved in eeprom. Grbl supports six work coordinate systems and two predefined positions via G28 and G30(set by G28.1 and G30.1). See the LinuxCNC gcode definitions for this. Grbl follows them very closely if not exactly for the ones it supports.

closed 2017-04-07 by chamnit

Incredible-IT, 2017-04-07:
See, again, you said, "Then you send this code." Period, end of sentence. Send what code?

That is what I'm trying to not do. Why home to the upper right then jog back to work origin and, reset? I want to use home position to orient the machine then automatically jog to my work origin. Otherwise homing is useless. I can just disable homing and limits and place the gantry and set wcs to zero.

How do I set my wcs origin point into the memory? Something like G10 L2 P1 X-350 Y-240 Z-27. 
I just tried it and got what I want. LinuxCNC.org is where I found my answer.

neilferreri, 2017-04-07:
@Incredible-IT Without homing there are never any absolute positions on your machine.  Homing allows the machine to tell you exactly where it is.  It really doesn't matter if you home to back right or front left, you are simply getting a repeatable starting point.  If you disable homing, 0,0,0 will be wherever the tool is when you power on.  In many cases, this would be ok.  If you will be running multiple operations on the same piece, changing tools, repeating jobs, using jigs, have permanent fixtures, etc., homing is a valuable step.  I tend to cut a lot of different jobs from the same stock with the same bit.  I can use a G10 command (as you read about on LinuxCNC) to set my Z0 once, and I don't have to worry about probing or jogging to the stock to set that value again.  On my main machine, I use a G28 for a front left easy tool change location, and I use a G30 location for a get the tool out of my way so I can admire my work at the end of a job.

Incredible-IT, 2017-04-07:
Thanks, so yes everything you say is true. My problem was there is a lot of opinion out there with no or little direction. The LinuxCNC docs are about the only concrete answers I have found. Last night I set my origin with "G10 L2 P1 X-302 Y-267 Z-28". That will permanently offset my machine "home" to my work "origin". If I need to temp set a new work origin I will use the G92 command. As you say, I can nest several items on one piece of stock, I will use G30 for my tool change so I jogged my pendant to that location and did a G30.1 command to set that point. I will prob use G28 for my safe location. Everything is working well now.

Now I just need to find a easy repeatable way of changing paint pens.

gerritv, 2017-04-07:
A good book on CNC machining, e.g. the ones by Peter Smid, will provide all the learning you need, check your local library. Or a site such as http://www.cnccookbook.com/MakeCNCParts/CNCMachineSetup.html. 
Generally Google is not your friend when learning something complex, it is great when looking up something specific.

chamnit, 2017-04-07:
@Incredible-IT : Both you and @gerritv are right. The Grbl documentation is somewhat lacking in how to use it, especially the homing documentation. That's being worked on and I added a note to add a general overview of how to use it and why, along with providing a clear link to the relevant LinuxCNC documentation. However, CNC isn't something new. It's been around for several decades since the first punch card computers. There are a wealth of resources available to learn how to CNC. Grbl kind of assumes that the user has some prior knowledge and is able to search out on their own. I'll make sure that some of the lower hanging fruit gets cleared up, since a lot of it isn't explicitly documented.

wlaw, 2017-04-07:
@Incredible-IT
I had the same type of problems.

this is what I did.
https://youtu.be/dRl4OHCeuMo

The video and text show how to call positions using machine quordinates.

Regards

Wlaw

vMeph, 2017-04-07:
@wlaw very nice, is that switch doing the same as a probe but with out a plate?

wlaw, 2017-04-07:
Thats correct.

I use the fixed switch point to calculate the height of the table.

Then set Zero at the table height.

You'll notice that I use G53 X-704.4 Y-1463.7 at the end of this macro
below.

This takes my machine to a designated spot based on machine coordinates and
sets them to X0 and y0.

But the machine coordinates are set with the "*Homing of the machine"*.

In the end I have a very repeatable return to a location.
More important, when I change a tool I no longer have to move the machine
to the stock and re-zero Z.
This automatically changes the *tool height offset* for me.
but best of all, all of the tabs I leave to keep the cut part in place are
in tact.

Now I enter the height of the stock in the software and let it cut.

This is the macro I use with Grbl Panel by @gerritv
G90
G21
$H
G92 Z0
M00
F2000
G01 Z-90
F1000
G38.2 Z-170
G92 Z0
G01 Z3
F100
G38.2 Z-3
G92 Z12.1
G01 Z75 F2000
G53 X-704.4 Y-1463.7
G92 X0 Y0

I have now changed the Z-switch to a more accurate slide, and prox. sensor
that I made.

My tip on my tool repeats within .02mm and that's close enough for me.
but the switch was also accurate within .07 mm. and that's close enough for
most.

You you want to see that I can record and update.
So far my video only has about 2500 hits in 1 year.