gnea/grbl #157: The voltage in laser mode

YExh00, 2017-03-15:
The maximum voltage of my laser engraver is 12v. And I don't know how to change the configuration in grbl of the maximum voltages from 5v to 12v. Could anyone help me?

cncpaw, 2017-03-15:
grbl can't do this. grbl's speed output is a PWM output from the arduino's AVR microcontroller chip, and that output is either 0V or 5V.
What you need is an external circuit and that circuit will depend a lot on exactly what laser it is driving and the available power supply. If you can post some details of both I might be able to help.

dungjk, 2017-03-15:
Grbl laser output pin voltage is between 0v and 5v (PWM mode). You should
you an external TTL circuit (you may build a simple one with mosfet or
transistor).

chamnit, 2017-03-15:
@YExh00 : Please don't post in multiple locations. It wastes everyones time having to answer in both places.

closed 2017-03-15 by chamnit