gnea/grbl #160: Unexpected reset on opening of serial connection.

doenerkebap, 2017-03-21:
I'm trying to interface grbl with a little python script.
But when I open the connection, grbl seems to reset , which is unexpected.
When connecting to grbl via HTerm, I can connect and disconnect without problems.

```
import serial

# initialize serial connection
ser = serial.Serial(None, 115200, timeout=2)
# specify port separately to prevent auto-connecting
ser.port = 'COM3'

# open and print welcome message
ser.open()
welcome_message = ser.readlines()
print(welcome_message)

# close
ser.close()

# now open again
ser.open()
welcome_message = ser.readlines()
print(welcome_message) # <- im expecting this message to be empty.
# but apparently grbl reset, lost all coordinates, and shows welcome message again!
```

chamnit, 2017-03-22:
This is normal behavior for Arduinos. They reset upon reconnect.

closed 2017-03-22 by chamnit

doenerkebap, 2017-03-22:
That's interesting, because I can connect and reconnect with HTerm without reseting the Arduino.

chamnit, 2017-03-22:
HTerm likely does not disconnect completely. There are plenty of online resources that explain the reset upon connect if you do a quick search.

doenerkebap, 2017-03-22:
I found the solution:
[https://github.com/pyserial/pyserial/issues/124](url)

Adding a simple
```
ser.dtr = None
ser.open()
```
before the open call prevents the Arduino to reset.

swarfer, 2017-03-22:
you can also prevent the Arduino reset by putting a 10uF electrolytic between the GND and RESET pins.
This means it will never reset via commands from the USB port so you have to press the reset button when flashing.