gnea/grbl #161: Unexpected Feed Rate Limit - Laser Mode - M4

AustinSaintAubin, 2017-03-22:
[Grbl 1.1f latest build]
Linear Move (G1) seems to top out at **2132** Feed Rate...
Feedrate for gcode its set to F4000.
[Laser Mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode) is enabled. ($32=1)

Increasing the Feed Rate Overrides have no effect.
Decreasing the feed rate overrides has no effect 
I am using CNCjs. v1.9.0-beta.1
[image]

As a side note, scans performed at 45 degrees do not seem to have this issue / the issue Horizontal scans do. 

G-Code Sample [Horizontal] (the one that is having the feed rate limited to 2132 issue)
```
 ; File: Dr-Evil-dr-evil-19031166-1280-1024_bw.bmp 
; 03-21-2017 at 23:27:58 
; PicLaser v. 2.1.2
; UNREGISTERED DEMO VERSION 
; AUTOMATIC SKIP LINES ENABLED 
; GRBL Gcode 
; FeedRate = 4000 
; Pixel Resolution = 0.1800 
; Max. Laser Value 255.0000 
; Min. Laser Value 42.0000 
; Engraving Angle = Horizontal 
M7
G21
F4000
M5
G01 X0.00 Y0.00
M03
X0 Y0 M4 S42
X0.18 Y0 M4 S42
X0.36 Y0 M4 S42
X0.54 Y0 M4 S42
X0.72 Y0 M4 S42
X0.9 Y0 M4 S42
X1.08 Y0 M4 S42
X1.26 Y0 M4 S42
X1.44 Y0 M4 S42
X1.62 Y0 M4 S42
X1.8 Y0 M4 S42
X1.98 Y0 M4 S42
X2.16 Y0 M4 S42
X2.34 Y0 M4 S42
X2.52 Y0 M4 S42
X2.7 Y0 M4 S42
X2.88 Y0 M4 S42
X3.06 Y0 M4 S42
X3.24 Y0 M4 S42
X3.42 Y0 M4 S42
X3.6 Y0 M4 S42
X3.78 Y0 M4 S42
X3.96 Y0 M4 S42
X4.14 Y0 M4 S42
X4.32 Y0 M4 S42
X4.5 Y0 M4 S42
X4.68 Y0 M4 S42
X4.86 Y0 M4 S42
X5.04 Y0 M4 S42
X5.22 Y0 M4 S42
X5.4 Y0 M4 S42
X5.58 Y0 M4 S42
X5.76 Y0 M4 S42
X5.94 Y0 M4 S42
X6.12 Y0 M4 S42
X6.3 Y0 M4 S42
X6.48 Y0 M4 S42
X6.66 Y0 M4 S42
X6.84 Y0 M4 S42
[... 815 lines cut ...]
X64.98 Y0.18 M4 S182.3294
X64.8 Y0.18 M4 S182.3294
X64.62 Y0.18 M4 S184
X64.44 Y0.18 M4 S184
X64.26 Y0.18 M4 S182.3294
X64.08 Y0.18 M4 S184
X63.9 Y0.18 M4 S191.5176
X63.72 Y0.18 M4 S190.6824
X63.54 Y0.18 M4 S184.8353
X63.36 Y0.18 M4 S182.3294
X63.18 Y0.18 M4 S179.8235
X63 Y0.18 M4 S177.3176
X62.82 Y0.18 M4 S175.6471
X62.64 Y0.18 M4 S175.6471
X62.46 Y0.18 M4 S178.9882
X62.28 Y0.18 M4 S178.1529
X62.1 Y0.18 M4 S176.4824
X61.92 Y0.18 M4 S173.9765
X61.74 Y0.18 M4 S176.4824
X61.56 Y0.18 M4 S176.4824
X61.38 Y0.18 M4 S177.3176
X61.2 Y0.18 M4 S177.3176
X61.02 Y0.18 M4 S177.3176
X60.84 Y0.18 M4 S175.6471
```

chamnit, 2017-03-22:
You are either running into the performance limit of the Arduino or have a max rate setting around 2100 mm/min. It's likely the former, as you are sending and executing about 200 blocks per second. I've heard Grbl can do more like 400-500 blocks per second in best case scenarios, where the sender is optimized for running laser raster jobs. 

Diagonals tend to run faster because the distance between blocks is 141.4% longer. 

To see an effect on feed overrides, you have to decrease the override such that the programmed rate x override is below the current rate. You likely did not decrease it enough.

closed 2017-03-22 by chamnit

AustinSaintAubin, 2017-03-22:
You seem to be correct, when I decrease the resolution from 0.18 to 0.4 I am able to hit my expected feed rate.

chamnit, 2017-03-22:
Sure thing. You can improve performance slightly if you remove unneeded commands like a Y that doesn't change, M4 commands after its first set, round your spindle values to an integer, and remove white space. This will reduce the serial RX overhead by half.

mayhem2408, 2017-03-22:
I run a 5 Watt laser running feeds for 1500 to 2000 at resolutions of 0.05 to 0.1. GCode is heavily optimised. I also recently switched over to using grbl-mega, with a serial buffer twice the size and the planner at 4 times the size. Runs smooth. I found a couple of senders could not keep up, but the recent version of cncjs has no problems.
@AustinSaintAubin What power laser are you running to get good burns at Feeds of 4000?

AustinSaintAubin, 2017-03-22:
@mayhem2408 I am using an [X-Controller](https://www.inventables.com/technologies/x-controller-kit) on my [X-Carve](https://www.inventables.com/technologies/x-carve). For the laser I am using an DIY 6W w/ NUBM44 450nm Laser Diode at the moment. I am transitioning into a [Opt Lasers PLH3D 6W](http://optlasers.com/en/plh3d-6w/175-plh3d-6w-5902693110637.html) that uses the same diode and lens, but a lot better driver and cooling(better housing). Both will/do make for quick engraving work with a 3 element lens, and quicker work with a G2 lens if I don't need the fine focus of the 3 element lens.
**What are you using to optimise your GCode?**

@chamnit I will work to optimes my GCode and [my optimiser web app](https://github.com/AustinSaintAubin/gcode-depths-to-shades).