gnea/grbl #162: Jogging Started by not able to Cancel

zentoolworks, 2017-03-23:
I can send a Jog command "$J=x500 F100" and it can move the motor as expected, but when I tried to cancel it by sending over a single char '\u0085', nothing happened, until it reached X500 and status turned back to "Idle". Any idea?

chamnit, 2017-03-23:
@zentoolworks : Hi Xin. It should work, but if it doesn't respond at all, it is usually caused by not sending a '0x85' value correctly. I would first verify whether or not you are sending the byte with an oscilloscope or some other means.

zentoolworks, 2017-03-23:
I am using the same method sending over "0x18" for soft reset and it worked, but 0x85 takes no effect. I will try to test in some other ways, oscilloscope may not be an option for me :(

langwadt, 2017-03-23:
I suspect unicode trouble, afaict with unicode anything above 0x7f turns into multi byte characters
so \u0085 is actually 0xC2 0x85

zentoolworks, 2017-03-23:
worked using this byte code: '\x0085' :) I am closing this ticket. Thank you, all.

closed 2017-03-23 by zentoolworks

109JB, 2017-03-24:
For others that may find this thread and happen to be programming in visual basic. In Visual basic you have to change the encoding used by the serial port to have extended ASCII characters work correctly. In my GUI code I have the following line to change the encoding of the serial port.

SerialPort1.Encoding = System.Text.Encoding.GetEncoding(1252)