gnea/grbl #165: Question about EEPROM write

109JB, 2017-03-31:
I am using [VER:1.1f.20170324:] [OPT:V,15,128] with echoes turned on. This is on an Arduino UNO clone with the 16U2 USB-serial chip.

While working on my GUI and in particular the code for updating grbl EEPROM settings, I have come across a problem that I believe is related to EEPROM write delays. The way my code works is it reads each grbl setting from a table and if the user has changed the value (compared to the current EEPROM value) it sends it to grbl using the send-response protocol. What is happening is that some lines sent are being truncated. For example on one try, the lines that were sent were:

$101=315
$102=315
$103=315

The echoed responses with "ok" and error responses were:

[echo: $101=315]
ok
[echo: $102=315]
ok
[echo: $103]
error:3

If many settings are changed, there might be several lines truncated.  In many tries, the lines that are truncated are entirely random, as are the number of character truncated. If I add an artificial 50 mSec delay after getting the "ok" response then everything works fine. I believe what is happening is as follows;

I believe what is happening is that the GUI receives an "ok" response and sends the next line, but grbl starts the EEPROM write for the previous line in the middle of receiving the next one and results in truncating the line being currently received. So a couple questions to this:

1. Do you think what I have described is what is actually going on?
2. Is the "ok" response sent before the EEPROM write for that value begins
3. If the answer to question 2 is yes, then should it be that way? Or should the "ok" come after EEPROM write is complete.
4. Is there a way to know when the EEPROM write is completed?

chamnit, 2017-03-31:
@109JB : 
1. Not exactly sure whats going on, but I think it might have something to do with the 'echo:' or how your GUI handles the echo message. I frequently stream settings to quickly configure Grbl for debugging a reported issue. I've never had a problem with using the Python streaming script when invoking the '-s' simple stream flag. If you still have this problem without echo enabled, try the Python streaming script. If you can reproduce this behavior there, then send me your streamed file, build info, and settings.

2. No. An EEPROM write is blocking and will only send an 'ok' response after it has been completed. You can see this in setting.c.

3. N/A

4. When you receive the 'ok'.

109JB, 2017-03-31:
Interesting. I will delve deeper and see what I can find.

doppelhub, 2017-05-03:
I've experienced this same issue since at least 0.8... multiple consecutive parameter writes leads to errors.  We get around this by using the $RST=* command to reset to our factory defaults.  Before "$RST=*" came along (1.0?), we used a basic write/read/verify to ensure each parameter was in fact set correctly.  Our first pass write rate was about 95%... which means there's a 65% chance of having a write error after just 20 parameters are written.  

Our protocol was different than yours, as we simply rewrote everything everytime (i.e. we didn't first check to see if the parameters were different).  We noticed that the error rate was drastically lower when the parameters to write were equal to what was already written.  I'm not sure why that's the case, but in conclusion, we'd typically have to write all the data 2 or 3 times before it was all written correctly.

As I mentioned above, we stopped bulk writing parameters when the $RST=* command came along, although this does require you to set all parameters before compiling.

It's possible the underlying parameter write bug I just reported is related.