gnea/grbl #171: Wiring a STOP button

jahnj0584, 2017-04-14:
Is there a way to hardwire a switch to send a STOP command? Thinking of my E-stop button, which has NO AND a NC terminal pair on it. When you close the switch, it opens the circuit (ie: Power in to the drivers) and closes the NO circuit, so my idea is that path is closed and would send the signal to the arduino/GUI to stop sending commands. The power shut off will physically stop the machine. After the switch is flicked back to normal, the GUI could just be in an alarm state and require a set up.

chamnit, 2017-04-14:
In the past, I've recommended to wire the E-stop to the Arduino reset pin (not Grbl's soft-reset). This should place the micro controller into a suspend state until the e-stop is released, whereby Grbl should boot immediately after. I've never done this myself, but theoretically it should work.

tklus, 2017-04-14:
In my mind, e-stop should be hit when bad things are going to or have already happened... I have my estop connected to my 24v power supply that is powering my steppers. This way, all motion stops immediately.  

I then have a "hold" button connected to the arduino that will stop motion but maintain power to the steppers. I can't remember what pin off the top of my head but there is one for slide hold or pause.

vMeph, 2017-04-14:
i have set on my box a button that power my 24v to connect all system, on the same button power i have a relay that i have put on the NO side connected to the arduino reset button, so when i close the power button it will close all system and at same time will reset arduino alarms, so when i turn it back on i dont have to remove usb cables to reset arduino and remove alarms,seems to work perfectly,before everytime i had a alarm i would have to power down case and then remove usb from arduino to be able to reset to remove alarms, was a pain!!

jahnj0584, 2017-04-14:
My idea is that when I hit the switch, it opens the 48V circuit to my drivers (halting motion). The NO circuit then closes, triggering the Hold/stop/reset/whatever pin on the arduino, so i dont totally lose the position of the code.

109JB, 2017-04-14:
If you remove power from the stepper drivers you do lose position

jahnj0584, 2017-04-15:
yea thats why i want to instantly issue a hold command too. I'd be
restarting it from the start of the toolpath so thats not my concern.
Saving 45$ router bits is.

chamnit, 2017-04-15:
IMHO, an e-stop event should not be concerned with retaining position. It should do whatever it takes to ensure safety. If you rehome afterwards and have used work coordinate systems, you should have your position still. You just need to know where in the program to restart, which can be difficult and tricky. You have to know the gcode state precisely and set it prior to restarting mid program. The GUI should do this task for you but there aren't any that I know of that will do this properly.

closed 2017-04-15 by chamnit

jahnj0584, 2017-04-15:
Yea that's fine. I can just use a text editor and remove whatever has
already been cut. So how do i wire the reset pin properly?

vMeph, 2017-04-16:
if you have a emergency button with NC and NO conections,  you can wire the power supply to NC and the NO to arduino reset , so when you turn on emergency button to give power to the circuit the NC powers all circuit , and in case  you press emergency when something goes wrong the NC becomes open and cuts all power to circuit and the NO will close and arduino resets .
at same time when you are interacting with UGS and if a alarm goes off all you would need to do is turn off emergency and turn back on and all alarms are gone sense arduino have reset..

dont know how others have it set but for me this is just perfect

jahnj0584, 2017-05-13:
that does make sense, and  I'll probably do it that way. Seems ideal,
really. Cancel power source, and reset the drivers and GUI. I right the
machine coords of my work 0 when i start anyways. I havent had or plan to
have to e-stop any time soon, but better safe then sorry