gnea/grbl #172: Pxx dwell command values

Simplystupid, 2017-04-14:
Compatibility across different software, not really an issue as such.
The Techne CADCAM generating my gcode is using msec as value after P command. Eg P5000 not P5 on spindle start-up.
I use Candle to send file to Grbl on arduino.
The only opportunity to alter the Pxx value to seconds seems to be in the Grbl 'code.c' source file.
If I put a value /1000 in the P value assignment, will that clash with any other code checking for divide by zero/non-integer etc?
` /* Non-Command Words: This initial parsing phase only checks for repeats of the remaining
           legal g-code words and stores their value. Error-checking is performed later since some
           words (I,J,K,L,P,R) have multiple connotations and/or depend on the issued commands. */
        switch(letter){
          // case 'A': // Not supported
          // case 'B': // Not supported
          // case 'C': // Not supported
          // case 'D': // Not supported
          case 'F': word_bit = WORD_F; gc_block.values.f = value; break;
          // case 'H': // Not supported
          case 'I': word_bit = WORD_I; gc_block.values.ijk[X_AXIS] = value; ijk_words |= (1<<X_AXIS); break;
          case 'J': word_bit = WORD_J; gc_block.values.ijk[Y_AXIS] = value; ijk_words |= (1<<Y_AXIS); break;
          case 'K': word_bit = WORD_K; gc_block.values.ijk[Z_AXIS] = value; ijk_words |= (1<<Z_AXIS); break;
          case 'L': word_bit = WORD_L; gc_block.values.l = int_value; break;
          case 'N': word_bit = WORD_N; gc_block.values.n = trunc(value); break;
          case 'P': word_bit = WORD_P; gc_block.values.p = value; break;
          // NOTE: For certain commands, P value must be an integer, but none of these commands are supported.
          // case 'Q': // Not supported`

The alternative is of course to keep opening gcode file in text editor.

Simplystupid, 2017-04-14:
gee, what happened to the formatting....'Pxx' is last value assigned in the code posted, so just read up from bottom.

chamnit, 2017-04-14:
The P word is interpreted as seconds for G4 per the LinuxCNC g-code standard (http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g4). Somewhere along the line, the hobby community altered this to mean milliseconds, which is not correct, and this spread throughout all of the hobby-class CAM programs. I would instead ask Candle to correct the G4 output or allow you to pick it in their settings.

closed 2017-04-14 by chamnit

gerritv, 2017-04-14:
@Simplystupid You asked this a while ago on the Candle github and got the same answer. The problem is with 'The Techne CADCAM generating my gcode is using msec as value after P command. Eg P5000 not P5 on spindle start-up.', not with Candle nor the other senders and not with Grbl or Gnea. Either get Techne CADCAM to fix/enhance their implementation or take the Candle code and make the alteration yourself.

Simplystupid, 2017-04-15:
Candle is compiled with Qt. Creator. It is not going to be straightforward for me to install the IDE and support software to find and alter the relevant code in a big complex project, just to convert dwell msec to sec. Techne CADCAM is not currently under active revision. Grbl has the source code available, and I thought, the support from the community. Hence my question posted OP.

gerritv, 2017-04-15:
@Simplystupid Ok, then you could modify Grbl for yourself? Based on the comment for <case 'P'> I think it is safe to add the divide by 1000. Nothing to lose by trying.

109JB, 2017-04-15:
look in "motion_control.c", line 195

void mc_dwell(float seconds)
{
  if (sys.state == STATE_CHECK_MODE) { return; }
  protocol_buffer_synchronize();
  delay_sec(seconds, DELAY_MODE_DWELL);
}

Change the last line in the function to 
delay_sec(seconds/1000, DELAY_MODE_DWELL);

That will make all the G4PXXXX commands work in milliseconds.

Simplystupid, 2017-04-16:
Many thanks, problem solved now.
I need the Grbl variable speed spindle option to send signal to 2nd arduino with PWM ramp up/down and limit the 12v DC motor start/stop current.