gnea/grbl #173: Potentially bad code in protocol_exec_rt_system()?

terjeio, 2017-04-16:
line 327:

`system_clear_exec_state_flag((EXEC_MOTION_CANCEL | EXEC_FEED_HOLD | EXEC_SAFETY_DOOR | EXEC_SLEEP));`

should be changed to 

`system_clear_exec_state_flag(rt_exec & (EXEC_MOTION_CANCEL | EXEC_FEED_HOLD | EXEC_SAFETY_DOOR | EXEC_SLEEP));`

in order to avoid losing events?

chamnit, 2017-04-16:
@terjeio : I'm aware of this potential issue. It's actually been there for a couple of years at least without anyone noticing until now. However, it should be exceedingly rare that something would be missed, because these events have to occur within a a few microseconds of each other to not be processed. 

Although I really like your simple solution, I'm hesitant to commit to it, because I'm not 100% certain that there won't effect the state machine in some unexpected way. There would need to be a lot of testing at extremes to ensure any edge cases are found and dealt with.

jasoroony, 2018-03-30:
In considering this change to my own 32 bit version (that it seems will show issues more readily because of the increase in MHz) would it be appropriate to do as terjeio says, then immediately copy back into rt_exec = sys_rt_exec_state and "goto" a new label inserted just prior to the opening if test for those 4 items, to retry again?  (Yeah, no-one likes "goto" but I just want an opinion on if this covers any edge cases?)

Then again, while I see the potential for issues here, I don't see what edge cases may exist that are not already going to cause problems immediately after the line of code (via pesky "anytime" interrupts) referred to in the OP.

jasoroony, 2018-03-30:
(Actually, making the "if" a "while" instead removes the need for a "goto")

terjeio, 2018-03-30:
I have changed the the code for the read of the variable into an atomic read and clear in order to make it even more robust:

``
    if (sys_rt_exec_state && (rt_exec = system_clear_exec_states())) { // Get and clear volatile sys_rt_exec_state atomically.
        ...
``

See the fork on my github account for more details.

jasoroony, 2018-03-31:
That's a good solution.

Incidentally, having had a look at your account, you might be interested in https://github.com/oilulio/Cisolate (it's not mine, but the guy did fix a bunch of things for me - and it already produced good PCBs before that - the automated drill gcode is what got me hooked).