gnea/grbl #175: integrated CAM software  for GRBL CNC laser

mkeyno, 2017-04-19:
sorry to open ticket for my question ,  normally I've used inkscape to draw CAD , then use gcode extension to convert to gcode , and finally feed file by serial  gcode sender to the machine , its not hard for me but not  for newbie  or technician whom work with easy Chines laser cutter software.  That interface directly send CAD file to the machine . Is there any software like Chines laser cutter which is adapted merely for GRBL firmware ?

chamnit, 2017-04-19:
Try LaserWeb. It's free, very powerful, and supports Grbl.

closed 2017-04-19 by chamnit

swarfer, 2017-04-19:
I recently discovered that one can integrate Candle https://github.com/Denvi/Candle/releases into SketchUcam.
SketchUcam is a Gcode generator plugin for Sketchup.
https://github.com/swarfer/sketchucam/releases

So, you can do your CAD in Sketchup, Generate the Gcode with SketchUcam (within Sketchup) and then load that Gcode in Candle as the 'Gcode viewer' and send it directly from there to the GRBL controller.

One must just remember to close Candle before editing the drawing some more, and also before closing Sketchup (since it opens as a child process and closing the parent can close the child, not good if your job is still running).

mkeyno, 2017-04-19:
thanks you both for your advise but  I'm well aware for laserweb and work with but as mentioned  its not easy  as the not free professional CNC machine , which is directly send CAD file to machine
Laserweb still very poor to configure multi layer  definition for different cutting power

chamnit, 2017-04-20:
@mkeyno : Keep in mind that LaserWeb is free and being continually developed at a very rapid pace by a very healthy community. If you want to see it become easier and more user-friendly, you can always join their effort to make something that everyone can use. If you can't help with coding, feedback and testing is always a valuable resource that most can help with.

mkeyno, 2017-04-20:
:+1: sure , thanks for your repo any way