gnea/grbl #176: stk500_recv(): programmer is not responding

lazydancer, 2017-04-20:
I receive an error when trying to flash an Arduino Uno with grbl v1.1f

Tested with grbl library on Arduino Software and using .hex file on Xloader. Using default configuration file. Tried also on another computer with the same settings.

Thoughts are the issue could be the age of the board being Uno R2 vs Uno R3 or if the firmware is compatible.
 
BN: Arduino/Genuino Uno
VID: 2341
PID: 0001
SN: 64935343633351303212

Below is error readout from Arduino Software using grblUpload:

```
Sketch uses 30374 bytes (94%) of program storage space. Maximum is 32256 bytes.
Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega328p -carduino -PCOM3 -b115200 -D -Uflash:w:C:\Users\James\AppData\Local\Temp\arduino_build_522168/grblUpload.ino.hex:i 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 3.3
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\James\AppData\Local\Temp\arduino_build_522168/grblUpload.ino.hex"
avrdude: writing flash (30374 bytes):

Writing | ###############################################avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
```

Thank you

n0th1n6, 2017-04-20:
This actually an arduino issue and not grbl.

chamnit, 2017-04-20:
@lazydancer : This is most likely due to the older boot loader on your Arduino. Try re-burning the boot loader by using a spare Arduino as a programmer. A google search should lead you to the right Arduino.cc page to show you how to wire it.`

closed 2017-04-20 by chamnit