gnea/grbl #177: $20 - Soft limits, boolean not working?

patgadget, 2017-04-23:
Hi,
i am using GRBL in Laser Mode
i also have remove the homing sequence for the Z-Axis.
and i also have comment out the HOMING_FORCE_ORIGIN to have a 0,0 in my lower left corner.

My Homing sequence work find, but i can ask to go over my soft limit without any alarm?
any idea?

Thanks
Pat

chamnit, 2017-04-23:
@patgadget : Not sure what you are asking. The point of soft limits is to stop you from crashing your machine, so an alarm is the correct response. If you are asking how to jog without exceeding soft limits, Grbl v1.1 new jogging mode will not throw an alarm when you command a motion outside of it. It'll just return an error. This new feature is outlined in the documentation.

closed 2017-04-23 by chamnit

patgadget, 2017-04-23:
Thanks i found my problem, i reverse the X and Y in the $130 and $131 parameter.
Pat