gnea/grbl #178: Homing cycle freezes

kvdveer, 2017-04-23:
I'm in the process of adding limit switches to my CNC machine. I have verified that my steppers and switches are wired correctly; as they move correctly on G0 and G1 commands, and they trigger the hard limits.

***The problem***
When I enable homing, and trigger it with `$H`, my machine just freezes. None of the axes move, the serial goes silent. The steppers are 'singing' though (not sure how to describe that), holding the axes locked. The only way out is a hard reset.

I've deduced (by way of random `write_serial`), that the first call to `st_wake_up()` never returns. The interrupt-driven operation of stepper.c is beyond my ability to debug, though.

***My setup***
I'm using grbl1.1f on a woodpecker board, with dual parallel normally open switches on all of the axes. The configuration is pretty much standard, although I have disabled some of the reporting to make the code fit the Arduino nano.

chamnit, 2017-04-23:
@kvdveer : I can't help you if you have modified the code, since I don't know what you did to it. It's highly likely your mods broke something, because homing works fine. Also, provide your settings. It's likely it just misconfigured, like your homing rates are way too high.

However, Grbl will fit on an Arduino Nano, if you burn the Arduino Uno boot loader onto it. It's a simple process, and I've done it to every Nano I own. First take a second Arduino, wire it up as an ISP (google Arduino as ISP), and connect both to your computer. Just set the board type to Arduino Uno, select the Nano port in the Arduino IDE, and click burn boot loader. Now you have an extra 1KB of flash and Grbl will fit. Your Nano will behave like an Uno (and requires you to always select Arduino Uno as the board type from now on). 

Just be aware that upload can sometimes timeout on some Nanos. This is easily fixed by clicking the Arduino hard reset within a couple of seconds after the flashing starts.

kvdveer, 2017-04-23:
> I can't help you if you have modified the code, since I don't know what you did to it

I only changed `config.h` to exclude the `REPORT_FIELD_*` defines - those only affect reporting, not stepping, so I'm quite confident stepping and homing are not affected.

> Also, provide your settings. It's likely it just misconfigured, like your homing rates are way too high.

```
$0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=100.000
$25=200.000
$26=100
$27=1.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=2500.000
$111=2500.000
$112=2500.000
$120=90.000
$121=90.000
$122=90.000
$130=0.000
$131=-0.000
$132=-0.000```

I'll toy around with the Arduino Uno bootloader, so I can run GRBL unmodified.

vMeph, 2017-04-23:
Try only use the homing at 1 the hard limits set at 0 and see if it works if so then you might have a noise issue on the hard limits

chamnit, 2017-04-23:
Your max travel values are zero. Homing uses these values to set home position and determine how far to search for a limit switch.

chamnit, 2017-04-23:
Also you are running at 33kHz. Above the max supported 30kHz step frequency. Grbl can become unstable in certain scenarios. So if you run into problem or strange behavior, lower your step frequency, either by reducing your step per mm or max rate. The former is preferred.

closed 2017-04-23 by chamnit

kvdveer, 2017-04-24:
Thank you for your help. I have updated the homing wiki to include what I've learned from this thread.

I'm still puzzled by your stepping frequency remark. The settings wiki doesn't mention stepping frequency, and none of the numbers I pasted have a value of 30 or 33. Could you point to how you've come to that diagnosis, so I can update the documentation for it?

langwadt, 2017-04-24:
800steps/mm times 2500mm/min  divided by  60sec/min = 33333 steps/sec

kvdveer, 2017-04-24:
Thank you for your response. I've updated the wiki to explain this. Hopefully, future users won't bother the issues section any more.

chamnit, 2017-04-24:
@kvdveer : Thanks for taking the time to write up a new wiki page. I'll be moving and integrating this into the main configuration page. FWIW, there is an automated check for max step frequency, but it's disabled due to flash limitations. (Literally a few hundred bytes left for critical build configurations). I think the only place the 30kHz limit is stated is on the main page and front wiki page. Placing that disclaimer in the configuration page should help future users from making the same error.