gnea/grbl #179: Inverting limit switches should not disable internal pull-up

Harvie, 2017-04-26:
Hi,
i've understood that inverting logic of limit switches using $5=1 also leads in disabling internal pullup. I think this does not make any sense. You should physicaly read switch state same way for NC and NO switches. Only difference is that you should invert how you interpret that state in code.

I don't see how disabling internal pullup helps. Especially when most people go for NC setup because of noise, which can be partialy alleviated by using internal pullup.

**Please make inverting limit switch logic and disabling/enabling pullup a separate setting. I am sure people might want to set it independently.**

 Also there's no really need to use external resistor to prevent frying something. I don't know how this statement got into the documentation:

> NOTE: If you invert your limit pins, you will need an external pull-down resistor wired in to all of the limit pins to prevent overloading the pins with current and frying them.

it just doesn't make any sense... if connecting NC switch to arduino would fry something it would also be fried by NO switch as soon as it was triggered for the first time. I think you need to understand that there is no real electrical difference between NC and NO switches. the only difference between NC and NO is purely mechanical. they only differ in what mechanical motion makes them conduct electricity... but in the end all switches (no matter if NO or NC) are just two wires that sometimes touch and sometimes don't.

chamnit, 2017-04-26:
@Harvie : Inverting the limits via the $ settings does not disable the pull-ups. They can only be disabled via the macro in the config.h file. So you should have a complete set of options, just one is compile-time, since it's not common.

closed 2017-04-26 by chamnit

Harvie, 2017-04-26:
Then why i am instructed to use pull-down resistor, when there's already internal pull-up??? does not make sense to have both. please fix the documentation then.

zeevy, 2017-04-26:
pull-down resistors are required if you invert the limit pins ($5=1 means pins are normally low and low to high will trigger the change) and using NO setup, with one end of the switch connected to Vcc.

chamnit, 2017-04-26:
@Harvie : Please calm down. You're getting agitated over a wiki page, which anyone can help keep updated, and an open-source, free firmware. There is no need to make demands. In much older versions of Grbl, inverting the limit pins did disable the internal pull-up resistor. About two years ago, Grbl stopped doing that and moved the disable internal pull-up to a compile-time option. Obviously, the updates to the documentation to that particular setting was overlooked.