gnea/grbl #180: G3/G4/G5 standards?

thinkyhead, 2017-04-27:
I'm trying to determine the "universal" standard for these commands, in relation to both spindle and laser.

- Is `M3` always "full on" and takes no parameters?
- Is `M4 Sn` the only proper command for setting partial power?
- Does the `S` parameter take a range of 0-100 or 0-255?

Any guidance you can provide will be very helpful!

chamnit, 2017-04-27:
@thinkyhead : Start [here](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m3-m4-m5). Grbl follows the LinuxCNC standard very closely, except in laser mode. Then, study [the laser mode wiki](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode) for how its different. For the `S` parameter, your answer is [here](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm).

closed 2017-04-27 by chamnit

jahnj0584, 2017-04-27:
Linux cnc explains these all

thinkyhead, 2017-04-27:
Thanks! I'd've done more Googling, but client has me rushing here…

gerritv, 2017-04-27:
Do you want it right or do you want it fast is the answer they should get. Can't have both.

thinkyhead, 2017-04-30:
> Do you want it right or do you want it fast

Indeed. I have a Venn diagram for them.

109JB, 2017-05-02:
For a spindle, M3 is clockwise rotation, M4 is CCW, and M5 is spindle off.

Both M3 and M4 can be used with an S### command to represent spindle RPM. The S parameter can be on the same line, or if M3 or M4 are used alone they will both revert to the last S value.